volumio/Volumio2 #284: [beta4] Add cifs network drive: Path must not start with path

dragotin, 2016-03-18:
I was struggling to add a proper network drive to volumio 2 beta4. After long fiddling around I found out that the path must not contain a leading '/. Removing the slash made it work properly.

There should be some checks on the content of the entry fields with error indicators.

apinner, 2016-03-18:
I was lucky and actually figured this out fairly quickly. However, I agree that this needs to be better explained or checked in the UI.

volumio, 2016-03-18:
Yes absolutely... added on to do list...

Atrejoe, 2016-03-22:
In my case even without adding the /, the share is not mounted:

[image]

The share is on a Qnap NAS.

Can I see an error log anywhere?

xipmix, 2016-11-29:
To view the error log, you will have to login over ssh
```
% ssh volumio@volumio
$ sudo journalctl
```
However I think I have addressed this in #877 .

xipmix, 2017-02-25:
This should be fixed now, in the next release after 2.041. Please close this when it works for you.

closed 2017-06-02 by volumio