volumio/Volumio2 #285: [beta4] Encoding issue on Playback View

dragotin, 2016-03-18:
German umlauts are not properly decoded on the Playback page:
[image]

The file comes from a mounted smb storage. In the Browse view, it says properly *Panik Präsident* as it should be so I assume the encoding of the filename is fine.

balbuze, 2016-03-18:
have a look to check tag of the track. I had such problem, and I found it was the tag that contain such problem (filename was ok)

dragotin, 2016-03-19:
Hmm, yes, good point. I am not sure how I can find out the encoding of the tags tough.

mp3info shows this:
```
File: 02-Cello.mp3
Title:   Cello                          Track: 2
Artist:  Udo Lindenberg
Album:   Panik Präsident               Year:  2003
```
which looks ok.

apinner, 2016-04-03:
@dragotin did you figure out this issue?

dragotin, 2016-04-03:
No, actually I think the mp3 file tags are fine and would suspect  a bug in the view.

apinner, 2016-04-04:
@dragotin I created a couple of mp3 files, one with the same name and tags as yours, and another that was just pure umlauts. They display absolutely fine for me, I think the issue is probably with your tags.

[image]

[image]

dragotin, 2016-04-04:
All right. Let's close this for now, thanks for investigation.

closed 2016-04-04 by dragotin