volumio/Volumio2 #293: Apple Airplay Network Name not changing

muzbox, 2016-03-21:
Hi

I have multiple Volumio 2 devices, when i change the name of the device the name of the airplay device stays as Volumio. The name of the device changes ( web, etc)

This is on RPi 3, version 0.856

Great work, keep it up

muzbox, 2016-03-21:
My mistake, a full power off and on and it now is displaying.

It would be good if it could change without the hard reset

volumio, 2016-03-21:
Yep, agree! I'll take a look at it!

apinner, 2016-04-03:
Did you get to look at this @volumio i can't find a commit

XiozTzu, 2016-05-11:
Very annoying to have this great player that looks very polished but when you go to view listings in Airplay all are in lowercase.  To me that just seems janky.

volumio, 2016-05-11:
We'll look at this, and decouple Airplay name from hostname

xipmix, 2016-11-05:
Something weird is going on here. I changed the player_name item  in [Hamburger]->System (saveGeneralSettings) a few times, ending up with it set to 'mumbo-jumbo'. I did a fresh boot and changed it again, to the unlikely name of 'a A a' (to test various bits of code).

The time stamps are not all exact, a ~ means approximately, or 'by this time'.
```
 22:34:13 onVolumioStart gets playerName mumbo-jumbo, uses it to set system.name
~22:34:31 (something) writes (something) to config.json.                       
 22:34:51 playerName is mumbo-jumbo in config.json                             
 22:36:06 getUIConfig says playerName is mumbo-jumbo                           
 22:36:49 change the name to 'a A a' in the web ui (saveGeneralSettings).
          player_name is 'a A a', newhostname a-a-a                               
          config.json does not change(? not 100% on this),
          but /etc/hostname does change (contains a-a-a).         
 22:37:04 setConf shows player_name mumbo-jumbo     
 22:37:05 playerName is mumbo-jumbo in config.json                             
~22:29:36 reboot                                                               
~22:40:10 before volumio starts, config.json has mumbo-jumbo, mtime same as before reboot
 22:40:20 onVoumioStart gets playerName mumbo-jumbo, uses it to set system.name
~22:42:16 volumio startup sound
 22:42:07 config.json changes, playerName is now a-a-a
 22:42:07 setConf shows player_name as a-a-a
```

I see at least two problems with this sequence of events.
 * At 22:37:04 setConf should be showing the new name, yes? And config.json should be getting updated at this point.
 * something magic happens to make the config.json file update during startup - after lots of poking around the code and on the filesystem I can't see how that datum could have been stored across the reboot. Which suggests something is getting stored in the browser session and retrieved at startup?

closed 2017-06-02 by volumio