volumio/Volumio2 #296: Deleted network drives are still displayed

Atrejoe, 2016-03-22:
```browse > music library > nas ``` also displays deleted network drives

apinner, 2016-04-03:
I can't recreate this issue, can you provide some more information or even a screenshot?

Atrejoe, 2016-04-04:
This was in the previous version. In the current-prerelease version I have a similar issue: right now with all shares deleted (and some time aited, and rebooted) it still shows some songs:
[image]

And so does my 'music library'

[image]

When adding the share again (under another name)

[image]

They are both there:

[image]

apinner, 2016-04-04:
Thanks for the info.

Atrejoe, 2016-04-04:
After leaving it a while (indexing the new location) the old location disappeared automagically.

(I now attempted to rename the alias for the share back to 'Hornbill' and no error was displayed, no confirmation either, and the alias was not persisted)

apinner, 2016-04-04:
Are you saying it is resolved?

Atrejoe, 2016-04-04:
Yes, but quite unclearly so. When a share is unmounted, you'd expect it to disappear immediately from your library.

volumio, 2016-04-04:
No, it takes some time: mpd db needs to be updated

closed 2016-04-04 by volumio