volumio/Volumio2 #305: Specified mount option nounix not used when mounting cifs share

poker335, 2016-03-26:
Volumio2 RC1, rpi B.

Hi, I specified the custom cifs mount option "nounix" in the volumio2 webapp for my cifs share (Advanced settings->Options). However, the share is still mounted with the conflicting "unix" option. The nounix option is ignored, although it is stored in the respective /etc/data/.../config.json file.

skikirkwood, 2016-03-28:
I too was bothered by this as I could not mount my SMB shared on my iMac, so I traced down the bug to app/plugins/system_controllers/networkfs/index.js where the options field was not being added to the fsopts variable before the mount call.  I tried to fix it with this modification, and it got farther - the mount said it was successful and it was listed in the Web UI, but the mounted status was "false".  If I knew how to debug a node.js app I'd look into this further, but this is how far I've got:

```javascript
ControllerNetworkfs.prototype.mountShare = function (shareid) {
	var self = this;

	var defer = libQ.defer();

	var sharename = config.get('NasMounts.' + shareid + '.name');
	var fstype = config.get('NasMounts.' + shareid + '.fstype');
	var pointer;
        var options = config.get('NasMounts.' + shareid + '.options');
	var fsopts;
	var credentials;

	if (fstype == "cifs") {
		pointer = '//' + config.get('NasMounts.' + shareid + '.ip') + '/' + config.get('NasMounts.' + shareid + '.path');
		//Password-protected mount
		if (config.get('NasMounts.' + shareid + '.user') !== 'undefined' && config.get('NasMounts.' + shareid + '.user') !== '') {
			credentials = 'username=' + config.get('NasMounts.' + shareid + '.user') + ',' + 'password=' + config.get('NasMounts.' + shareid + '.password') + ",";
		} else {
			credentials = 'guest,';
		}
		fsopts = credentials + "dir_mode=0777,file_mode=0666,iocharset=utf8";
                if (options !== 'undefined' && options !== '') {
                        fsopts = fsopts + "," + options;
                }
```

xipmix, 2016-10-22:
Is this still an issue in 2.001? If so, could you try adding to app/plugins/system_controller/networkfs/index.js 
```
  var mountpoint = '/mnt/NAS/' +  mountid
+ console.log('mounting "' + pointer + '" with fsopts="' + fsopts + '"');
```
and reporting what it says? The output will be in the system log (```sudo journalctl >system.log```)

It might be that the ```dir_mode``` and ```nounix``` options interfere with each other; my reading of the manpage says no but I could be wrong. Perhaps try doing a cifs mount manually, with the -v option, to see what is happening?

closed 2017-06-02 by volumio