volumio/Volumio2 #306: [rc1] perl: warning: Setting locale failed

ajongsma, 2016-03-26:
I noticed two minor issues with the locale configuration:
- locales isn't installed
- en_US.UTF-8 isn't available

I resolved the errors by:
- sudo apt-get install locales
- sudo dpkg-reconfigure locales
- Enable and select: en_US.UTF-8

volumio, 2016-04-15:
This adds lot of space to default install... Is there any other way to solve this?

xipmix, 2016-10-25:
I think I can see how to reduce the impact, by using localepurge. The (working) patch for raspberry pi is shown below. 

Installing locales uses up about 70MB, ~75 if you include manpages. Stripping back down to one locale saves about 72MB, so the net impact should be about 3MB. If this approach is acceptable I can prepare a PR for Build. 

But other than shutting up perl, what problem does this change solve?

```
diff --git a/recipes/arm.conf b/recipes/arm.conf
index a424a55..4d94427 100644
--- a/recipes/arm.conf
+++ b/recipes/arm.conf
@@ -24,7 +24,7 @@ keyring=debian-archive-keyring
 suite=jessie
 
 [Utils]
-packages=nano psmisc ethtool curl ca-certificates ntp git-core parted wget ava
hi-daemon avahi-discover libnss-mdns zsync udisks udisks-glue policykit-1 less 
fake-hwclock usbutils
+packages=nano psmisc ethtool curl ca-certificates ntp git-core parted wget ava
hi-daemon avahi-discover libnss-mdns zsync udisks udisks-glue policykit-1 less 
fake-hwclock usbutils locales localepurge
 source=http://archive.raspbian.org/raspbian
 keyring=debian-archive-keyring
 suite=jessie

diff --git a/scripts/raspberryconfig.sh b/scripts/raspberryconfig.sh
index 3f5fa43..75607d3 100755
--- a/scripts/raspberryconfig.sh
+++ b/scripts/raspberryconfig.sh
@@ -31,6 +31,25 @@ i2c-dev
 i2c-bcm2708
 " >> /etc/modules
 
+# Reduce locales to just one beyond C.UTF-8, to close #306 without costing 60M
B
+echo "Stripping back locales"
+# First, add the locales we want to keep to the config file
+echo "
+C.UTF-8
+en_US.UTF8
+" >> /etc/locale.nopurge
+# To remove existing locale data we must turn off the dpkg hook
+sed -i -e 's/^USE_DPKG/#USE_DPKG/' /etc/locale.nopurge
+# Ensure that the package knows it has been configured
+sed -i -e 's/^NEEDSCONFIGFIRST/#NEEDSCONFIGFIRST/' /etc/locale.nopurge
+dpkg-reconfigure localepurge -f noninteractive
+# Purge
+localepurge
+# Turn dpkg feature back on, it will handle further locale-cleaning
+sed -i -e 's/^#USE_DPKG/USE_DPKG/' /etc/locale.nopurge
+dpkg-reconfigure localepurge -f noninteractive
+
+
 echo "Alsa Raspberry PI Card Ordering"
 echo "
 options snd-usb-audio nrpacks=1

```

volumio, 2016-10-25:
That's a great idea! 
However this should be done both for arm and x86 recipes.
Then the locales stripping must be moved from raspberryconfig to line 10 of volumioconfig (or maybe just setting the default locale in the section above)
what do you think?

closed 2016-10-27 by volumio