volumio/Volumio2 #308: [rc1] Pi isn't discoverable in OS X via Finder > Shared

ajongsma, 2016-03-26:
Resolved this minor maybe not desirable feature by installing netatalk, maybe left out intentionally?

apinner, 2016-04-03:
Not sure I understand this issue, was 1.55 discoverable? 

Is this just to access the volumio shares on the Pi?

ajongsma, 2016-04-04:
1.55 wasn't discoverable either as far as I remember. It's a personal preference of mine to have the netbios name viewable while, if I'm not able to airplay for some reason, it's a very quick way to see if the pi is actually reachable.

apinner, 2016-04-05:
This just seems like it would add unnecessary elements to the build with little to no useful function. Considering you can just check the webUI or SSH.

closed 2016-04-05 by apinner

xipmix, 2016-10-27:
Bother. fatfingered the bug id. can someone please close this again?