volumio/Volumio2 #309: Kernel support for RT7601 USB WiFi

cshotton, 2016-03-28:
To help with support for newer, long range wifi dongles like http://www.amazon.com/Cable-Matters-Wireless-150Mbps-Detachable/dp/B00MXD2T5G?tag=leesburgtalk-20

Debian Jessie adds support for drivers for the RT7601-based wifi interfaces to be loaded, but they aren't part of the default distro. The drivers just need to be copied into the proper location. Here's the scoop:

http://chips.com.pk/2016/02/24/how-to-make-mt7601-wifi-adapter-work-on-raspberry-pi/

FishTest, 2016-04-02:
I think you can make your custom Volumio2 Image with offical image build script
https://github.com/volumio/build
git clone https://github.com/volumio/build --depth=1
nano scripts/raspberryconfig.sh
after this section:
# Kernel 4.1.18 for Pi3 Support
echo y | SKIP_BACKUP=1 rpi-update 6e8b794818e06f50724774df3b1d4c6be0b5708c

echo "Adding PI3 Wireless firmware"
wget https://github.com/RPi-Distro/firmware-nonfree/raw/master/brcm80211/brcm/brcmfmac43430-sdio.bin -P /lib/firmware/brcm
wget https://github.com/RPi-Distro/firmware-nonfree/raw/master/brcm80211/brcm/brcmfmac43430-sdio.txt -P /lib/firmware/brcm

add
wget https://github.com/porjo/mt7601/raw/master/src/mcu/bin/MT7601.bin -O /lib/firmware/mt7601u.bin

finally run the build script
you can get your custom image.

volumio, 2016-04-02:
Can you send a PR for this?

cshotton, 2016-04-05:
Current Raspi firmware updater (rpi-update) includes kernel 4.1.20-v7, which includes the specific driver needed for these devices. If you support this kernel version, this ticket is unnecessary.

closed 2016-06-27 by apinner