volumio/Volumio2 #310: Sleep Timer should stop music, not shut down device

cshotton, 2016-03-28:
The current Sleep Timer functionality is virtually useless. It actually shuts down the hardware device when the timer expires. This makes the alarm functionality impossible to use. That is, if you fall asleep to music and have the sleep timer turn it off, the device is powered down and the alarm functionality can never start the music again.

The desired behavior is for the sleep timer to expire and simply stop music playback. That way the mpd server can still be accessed without requiring a reboot of the host computer.

macmpi, 2016-04-04:
actually both usecases make sense: in some case you may want to actually power it down.
Maybe a toggle allowing either idle (stop music) or shutdown would make sense
Thanks!

volumio, 2016-04-04:
https://github.com/volumio/Volumio2/commit/12a91f117e0736dc30b27f676631cc0197f893fa

closed 2016-04-04 by volumio