volumio/Volumio2 #311: Soundcloud and YouTube

nibelungen, 2016-03-30:
Hey,

at first thanks for the great work so far on volumio2.
Maybe some one can clarify the current state for the integration of soundcloud 'urls' (supported by mpd since 0.17 and got updated to use https since 0.19) and of youtube 'urls' (maybe with youtube-dl).
Is this on the roadmap of volumio2 too?

Best regards,
-andreas

apinner, 2016-03-30:
Aren't both of those low quality with very compressed audio? I'm not sure that fits with the whole 'audiophile' idea does it?

silopolis, 2016-03-31:
@AlphaPapa may be (not even sure), but "what can do more, can do less", and
many of us may not want/be able to have several systems depending on media
quality !

apinner, 2016-03-31:
That's fair enough, just playing devils advocate really.

alphapapa, 2016-03-31:
@silopolis What's that you say?

nibelungen, 2016-04-01:
@apinner Where does low and high quality start and end? Currently i stream via bubbleupnp and mycloudplayer soundcloud urls to an volumio rpi, output via audiojack (in my garden) So i really dont care in this case for high (hifi) quality... Just wanna hear music from different sources, mostly soundcloud at the moment.
So for me - and maybe a lot of other volumio users, it would be a nice feature to use the mpd built in option to play soundcloud urls.

volumio, 2016-04-01:
To me is fine to add soundcloud... What would be the browsing mechanism for it that you envision?

apinner, 2016-04-01:
Like I said, just playing Devils advocate! I don't mind if either are added..

Soundcloud might be tricky, their API has rate limits on play requests of 15,000 per day per app. Like you say, its hard to imagine a mechanism to browse a site like soundcloud within volumio. Maybe just the ability to search?

nibelungen, 2016-04-01:
@volumio Great to hear!
Maybe like the xbmc/kodi plug-in and as @apinner mentioned - just a search and list the results.
Or for a first shot - just a input field to place the soundcloud url (as far is i know, mpd just need the url).

nibelungen, 2016-04-08:
Maybe helpful http://jeroendoggen.github.io/youtube-to-mpd/

TechDark, 2017-01-29:
maybe the same layout just like the spotify browsing

omaralvarez, 2017-04-01:
Any news on this feature? I would love to be able to listen to SoundCloud liked tracks or playlists.

Zelaf, 2017-04-22:
If you need to log in to soundcloud to be able to browse it you can display artists you follow in a page, then when you enter artists that you follow you can click on songs etc.

Also, would love to see soundcloud.

sla89, 2017-04-24:
FYI: we are currently implementing a fully integrated youtube plugin for volumio.

sepastian, 2017-04-27:
@sla89, what's the status on this? Can it be tested already?

sla89, 2017-04-28:
@sepastian no, not yet. Do you want to test it when we have a stable development version ready like an alpha or beta?

sepastian, 2017-04-28:
Sure, let me know when I can test.

Zelaf, 2017-04-28:
If it's not an issue I could also test it. It would get me a chance to work on my bug reporting skills and I would also get more chances to using my Volumio setup :D

TechDark, 2017-04-28:
So there is a soundcloud plugin?

Zelaf, 2017-04-30:
@TechDark I dunno, is there?

sla89, 2017-05-01:
@sepastian and @Zelaf I will come back to you soon 👍 Thanks for your help.
@TechDark as far as I know there is no plugin for soundcloud

crisp00, 2017-05-08:
Hi everyone! 

The YouTube plugin that _sla89_ and _I_ have been working on has finally been pushed to beta!!! You can find it in your __Volumio UI__ ,  _Plugins_ ->_Search Plugins_ ->_Music Services_ ! 

For bug reports and requests the best place is the volumio forums, where we have an official thread. You can also post an issue on my repo  __PiYoutube__  if you prefer.

A SoundCloud plugin may be coming next! Good things happen to those who wait!

Best, Cris Pintea.

omaralvarez, 2017-05-09:
I'm eagerly awaiting for the Soundcloud plugin. Godspeed and thanks my friend!

volderette, 2017-06-01:
Soundcloud would be my most awaited feature as well.

BenjaminKauer, 2017-10-27:
Soundcloud support shouldn't be that much of a problem, if we make the API key configurable by the user. It'd be the users job to receive a Soundcloud API key (which is currently not possible, btw).

See also: [forum post "SOUNDCLOUD SUPPORT"](https://volumio.org/forum/soundcloud-support-t3059.html)

frederiktoft, 2018-12-13:
Just want to put out a request for Soundcloud support as well. Mopidy supports it with personal api key, so it should be feasible to do a Volumio SoundCloud plug-in using the same method.

dmnBrest, 2018-12-26:
+1 for Soundcloud

gruentee, 2019-01-16:
Yes, SoundCloud support would be really great.

w97, 2019-01-22:
please make this work, thank you! :)

Pejosonic, 2019-05-01:
I would be great to have a Soundcloud integration!!

mujahed1987, 2019-05-02:
Soundclound integration will be great addition to Volumio

hussong, 2019-05-16:
SoundCloud audio quality is pretty good with a SoundCloud Go+ account (10 €/mo), then you can stream most tracks in 256k AAC (some are still only 128k MP3, I guess this is because the original upload wasn't higher quality). It is also possible to download these streams with a patched version of youtube-dl. 

I would really enjoy a SoundCloud integration as well!

cmosek, 2019-11-27:
+1 for Soundcloud integration!

Pejosonic, 2020-02-12:
Hi, i really would love to find the way to listen Souncloud tracks on my Volumio player.
Is there some workaround to do this in the current Volumio version?

Is there a particullar reason why the team it's not taking place for this feature request?
Maybe we could take some inspiration for the browsing from the Soundcloud Kodi add-on?

Thank you!
Regards

Titaantje, 2020-05-14:
Soundcloud integration, is it there yet?? 
Even the most basic integration like an url plugion would be BOMB :)