volumio/Volumio2 #313: [rc1] Cannot add NAS folder

xxxbugxxxx, 2016-03-31:
Tried to add NAS folder and it shows scuessful however, nothing got saved.

apinner, 2016-03-31:
Can you post a bit more information or even better a screen shot? That information on its own isn't very useful!

apinner, 2016-04-11:
No response, reopen if necessary.

closed 2016-04-11 by apinner

vincentgalore, 2016-11-23:
Hi all. I'm afraid we are having the exact same problem with mounting and seeing network drives via the web UI / system_controller-networkfs tab. The issue is consistent, and remains even after multiple reboots, power cycling and factory reset of the unit. Attempting a fresh image now, and troubleshooting the network here.

What could be causing this and how do we resolve it?

xipmix, 2016-11-23:
What version do you see this on? ```(cd /volumio; git log -1)```
Can you post logs? You will need to ssh in and run 'journalctl >logfile', then post the file here.

vincentgalore, 2016-11-23:
I'm over on the forums as well; the last post is mine.
https://volumio.org/forum/can-see-network-drive-webui-t4643.html

We are running -
System Version: 2.001 Released: Sat Oct 15 16:29:11 CEST 2016

We try and post the log by running the command you list above, but the terminal says that 'no journal files were found.'

This is what the terminal posts when running 'sudo journalctl -f' on selecting 'My Music' from the settings sidebar and attempting to mount a network drive that is visible; no currently mounted network drives are displayed via the UI.

Volumio Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
volumio@volumio:~$ sudo journalctl -f
-- Logs begin at Wed 2016-11-23 21:11:22 UTC. --
Nov 23 23:54:08 volumio systemd[1605]: Starting Basic System.
Nov 23 23:54:08 volumio systemd[1605]: Reached target Basic System.
Nov 23 23:54:08 volumio systemd[1605]: Starting Default.
Nov 23 23:54:08 volumio systemd[1605]: Reached target Default.
Nov 23 23:54:08 volumio systemd[1605]: Startup finished in 41ms.
Nov 23 23:54:08 volumio systemd[1]: Started User Manager for UID 1000.
Nov 23 23:54:08 volumio sshd[1590]: lastlog_openseek: Couldn't stat /var/log/lastlog: No such file or directory
Nov 23 23:54:08 volumio sshd[1590]: lastlog_openseek: Couldn't stat /var/log/lastlog: No such file or directory
Nov 23 23:54:11 volumio sudo[1618]: volumio : TTY=pts/0 ; PWD=/home/volumio ; USER=root ; COMMAND=/bin/journalctl -f
Nov 23 23:54:11 volumio sudo[1618]: pam_unix(sudo:session): session opened for user root by volumio(uid=0)
Nov 23 23:54:17 volumio volumio[832]: info: CoreCommandRouter::getUIConfigOnPlugin
Nov 23 23:54:17 volumio volumio[832]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Nov 23 23:54:17 volumio volumio[832]: info: CoreCommandRouter::executeOnPlugin: networkfs , listShares
Nov 23 23:54:18 volumio volumio[832]: fail....TypeError: Cannot read property 'substring' of undefined
Nov 23 23:54:18 volumio volumio[832]: Failed getting mounts size { path: 'ergh-vs-regulate-18-03-11\t',
Nov 23 23:54:18 volumio volumio[832]: ip: '192.168.0.8',
Nov 23 23:54:18 volumio volumio[832]: id: '603b600e-2d2c-481d-9746-06bc609520b2',
Nov 23 23:54:18 volumio volumio[832]: name: 'ergh-vs-regulate-18-03-11\t',
Nov 23 23:54:18 volumio volumio[832]: fstype: 'cifs',
Nov 23 23:54:18 volumio volumio[832]: username: '',
Nov 23 23:54:18 volumio volumio[832]: password: '',
Nov 23 23:54:18 volumio volumio[832]: options: '',
Nov 23 23:54:18 volumio volumio[832]: mounted: false,
Nov 23 23:54:18 volumio volumio[832]: size: '' }
Nov 23 23:54:18 volumio volumio[832]: fail....TypeError: Cannot read property 'substring' of undefined
Nov 23 23:54:19 volumio volumio[832]: info: CoreCommandRouter::executeOnPlugin: networkfs , discoverShares
Nov 23 23:54:20 volumio smbd[1645]: [2016/11/23 23:54:20.777239,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Nov 23 23:54:20 volumio smbd[1645]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Nov 23 23:54:20 volumio smbd[1645]: [2016/11/23 23:54:20.783746,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Nov 23 23:54:20 volumio smbd[1645]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Nov 23 23:54:20 volumio smbd[1645]: [2016/11/23 23:54:20.784219,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Nov 23 23:54:20 volumio smbd[1645]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Nov 23 23:54:20 volumio smbd[1645]: [2016/11/23 23:54:20.809700,  0] ../lib/util/debug.c:603(reopen_logs_internal)
Nov 23 23:54:20 volumio smbd[1645]: Unable to open new log file '/var/log/samba/log.smbd': No such file or directory
Nov 23 23:54:21 volumio volumio[832]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Nov 23 23:54:25 volumio volumio[832]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Nov 23 23:54:26 volumio volumio[832]: info: CoreCommandRouter::executeOnPlugin: networkfs , addShare
Nov 23 23:54:26 volumio volumio[832]: info: Adding a new share
Nov 23 23:54:26 volumio volumio[832]: info: Checking key NasMounts.85bbd437-9896-491b-99e2-1be9731e0fcb
Nov 23 23:54:26 volumio volumio[832]: info: Checking key NasMounts.c82fd602-7719-4312-85b9-74f5fac94617
Nov 23 23:54:26 volumio volumio[832]: info: Found correspondence in configuration
Nov 23 23:54:27 volumio volumio[832]: info: CoreCommandRouter::executeOnPlugin: networkfs , listShares
Nov 23 23:54:27 volumio volumio[832]: fail....TypeError: Cannot read property 'substring' of undefined
Nov 23 23:54:27 volumio volumio[832]: Failed getting mounts size { path: 'ergh-vs-regulate-18-03-11\t',
Nov 23 23:54:27 volumio volumio[832]: ip: '192.168.0.8',
Nov 23 23:54:27 volumio volumio[832]: id: '603b600e-2d2c-481d-9746-06bc609520b2',
Nov 23 23:54:27 volumio volumio[832]: name: 'ergh-vs-regulate-18-03-11\t',
Nov 23 23:54:27 volumio volumio[832]: fstype: 'cifs',
Nov 23 23:54:27 volumio volumio[832]: username: '',
Nov 23 23:54:27 volumio volumio[832]: password: '',
Nov 23 23:54:27 volumio volumio[832]: options: '',
Nov 23 23:54:27 volumio volumio[832]: mounted: false,
Nov 23 23:54:27 volumio volumio[832]: size: '' }
Nov 23 23:54:27 volumio volumio[832]: fail....TypeError: Cannot read property 'substring' of undefined

xipmix, 2016-11-24:
On Wed, Nov 23, 2016 at 03:56:18PM -0800, vincentgalore wrote:
> I'm over on the forums as well; the last post is mine.
> https://volumio.org/forum/can-see-network-drive-webui-t4643.html

Ok thanks.

> We are running -
> System Version: 2.001 Released: Sat Oct 15 16:29:11 CEST 2016
> 
> We try and post the log by running the command you list above, but
> the terminal says that 'no journal files were found.'

Ah yes, I forgot to put in the sudo bit. You need root privileges
to read the system journal.

> This is what the terminal posts when running 'sudo journalctl -f'
> on selecting 'My Music' from the settings sidebar and attempting
> to mount a network drive that is visible; no currently mounted
> network drives are displayed via the UI.
> 

This is exactly the kind of thing that is needed, thanks.
However since version 2.001, things have changed quite a bit and
this issue may be fixed. Could you try downloading 2.031
http://updates.volumio.org/pi/volumio/2.031/volumio-2.031-2016-11-23-pi.img.zip
and test again? Probably best to burn this to an empty SD card
so you can switch back to 2.001 if you want.

vincentgalore, 2016-11-24:
Yes, excuse my error, I'm relatively new to the terminal side of things. Although even with the sudo prefix the terminal doesn't produce anything. I'll read up on that.

Another thread goes into a little more detail, whereby an invalid share throws the system out and cannot be recovered or rolled back. It is also particularly fussy with shared network folders - not a mapped networked drive - which appear as directories alongside their respective radio button post scan, but that appear to only successfully mount by chance; there is no pattern I can see. That is clearly a by-product of what we are experiencing once it occurs via plugin/system_controller-networkfs. 

https://github.com/volumio/Volumio2/issues/315

Same thing occurs in 2.031 which we've just updated to -
volumio@volumio:~$ cd /volumio; git log -1
commit 3aedc4d9179c081941d69f0b05f06d81ea07d2d8
Author: Ghembs <ggambacorta88@gmail.com>
Date:   Wed Nov 23 15:49:21 2016 +0000

vincentgalore, 2016-11-24:
Thanks for your assistance on this one, I know I'm not the only one suffering from this; the platform is still usable, of course, it just would be lovely to be able to edit successful / failed network mounts via the UI, rather than having it all just vanish.

xipmix, 2016-11-24:
> it just would be lovely to be able to edit successful / failed
> network mounts via the UI, rather than having it all just vanish.

So if I understand correctly, if a mount fails for some reason,
you get a warning popup (orange) or an error popup (red) and the
failed entry does not appear in the list of mounts, so you cannot
fix up the things that are wrong, but have to start all over?

From the forum post it seems there is another part to this -
the share is actually mounted if you check from the command line?

It would be helpful to know if the mount is actually complete or
there is just a mount-point directory left over after a failed
attempt. To show this, try running:
  $ df -Ph
This will show all the filesystems mounted, including networked ones.

To look for leftover mountpoints, which df will not show,
  $ ls -l /mnt
  $ ls -l /mnt/NAS

vincentgalore, 2016-11-24:
To my recollection I've never seen an orange warning message via the 'toast' modal pop-ups. Green on success, and all is good, and red when the system _decides_ to feedback that it has failed; it appears to e very selective. Often, upon hitting the 'save' key for a new mount which is likely to fail - either our end, or via some .js bug system end - there is no pop up or feedback of any kind. Then we refresh, and discover that all previous visible mounts w/ the pencil edit and trash icon via the UI have disappeared, and we can never recover from that situation; only a reformat and a fresh image cures it.

I'm pushing bits via sFTP, so I've had a quick look at the /mnt/NAS/ directory and do find the mount-point directory /mnt/NAS/ergh-vs-regulate-18-03-11, which as you say, is probably left over after a failed attempt; no data is stored there. Running:
  $ df -Ph
shows only '/mnt/NAS/media' is, which mounted without issue, and continues to push audio:
volumio@volumio:~$ df -Ph
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2  2.0G  497M  1.4G  26% /imgpart
/dev/loop0      247M  247M     0 100% /static
overlay         5.1G  738M  4.1G  16% /
devtmpfs        479M     0  479M   0% /dev
tmpfs           487M  8.0K  487M   1% /dev/shm
tmpfs           487M  4.7M  482M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           487M     0  487M   0% /sys/fs/cgroup
tmpfs           487M   12K  487M   1% /tmp
tmpfs           487M     0  487M   0% /var/spool/cups
tmpfs           487M   72K  487M   1% /var/log
tmpfs           487M     0  487M   0% /var/spool/cups/tmp
/dev/mmcblk0p1   61M   29M   33M  47% /boot
//ASUS/media    465G  448G   18G  97% /mnt/NAS/media
tmpfs            98M     0   98M   0% /run/user/1000

The leftover mount-point commands generate this, respectively:
volumio@volumio:~$ ls -l /mnt
total 4
lrwxrwxrwx 1 root root   14 Nov 23 16:06 INTERNAL -> /data/INTERNAL
drwxrwxrwx 1 root root 4096 Nov 23 21:13 NAS
lrwxrwxrwx 1 root root    6 Nov 23 16:06 USB -> /media
volumio@volumio:~$ ls -l /mnt/NAS
total 16
drwxr-xr-x 2 volumio volumio 4096 Nov 23 20:50 Final
drwxr-xr-x 2 volumio volumio 4096 Nov 23 20:48 IVYWARE-PC
drwxr-xr-x 2 volumio volumio 4096 Nov 23 21:13 ergh-vs-regulate-18-03-11?
drwxrwxrwx 2 root    root    4096 Sep  9 04:17 media

The only  pattern I see is regarding permission settings, but our IVYWARE-PC network mount is available when the workstation is on.

xipmix, 2016-11-24:
Looking at your log output more carefully, I see a '\t' at the end of the share path. That's a tab, ie a whitespace character. Is that supposed to be there? It's known that whitespace in the NAS path can cause problems (see #856) and I wonder if that is the root of your problem here. What happens is the code that tries to get the share size fails (it calls 'df' and fails to parse the output in this case) and I don't know if that failure is handled properly. It may exit from the 'add a share' routine in a strange way.

I have a pull request for handling that issue waiting, but it needs review by someone who actually understands javascript. While we're waiting for that can you try the following:
 * delete the ergh-vs-regulate-... share, in the web UI, if you can
 * log in and ```sudo rmdir /mnt/NAS/ergh-*```  (not strictly needed but won't hurt)
 * try to create the share again, taking care to type the path without any tabs or spaces
 * paste the journalctl -f here.

vincentgalore, 2016-11-25:
That's lovely, have just got in. Will turn everything on and have a look. We really appreciate your help on this.

vincentgalore, 2016-11-25:
Ran the commands as above, and now the system just hangs. Can find it via Angry IP, but cannot reach the web UI at tall. Funnily enough I can still access via sFTP, but when I try to view the /mnt/NAS directory the connection times out; all other directory listings are successful. Will flash a new image and start afresh tomorrow.

Via Filezilla -
Error:	Connection timed out after 20 seconds of inactivity
Error:	Failed to retrieve directory listing

Log -

Nov 25 00:07:25 volumio kernel: sftp-server     D 805b3864     0  1267      1 0x00000000
Nov 25 00:07:25 volumio kernel: [<805b3864>] (__schedule) from [<805b3ddc>] (schedule+0x50/0xa8)
Nov 25 00:07:25 volumio kernel: [<805b3ddc>] (schedule) from [<805b41b4>] (schedule_preempt_disabled+0x18/0x1c)
Nov 25 00:07:25 volumio kernel: [<805b41b4>] (schedule_preempt_disabled) from [<805b5874>] (__mutex_lock_slowpath+0xac/0x164)
Nov 25 00:07:25 volumio kernel: [<805b5874>] (__mutex_lock_slowpath) from [<805b598c>] (mutex_lock+0x60/0x64)
Nov 25 00:07:25 volumio kernel: [<805b598c>] (mutex_lock) from [<7f281cd0>] (cifs_reconnect_tcon+0x84/0x320 [cifs])
Nov 25 00:07:25 volumio kernel: [<7f281cd0>] (cifs_reconnect_tcon [cifs]) from [<7f28200c>] (smb_init+0x30/0x78 [cifs])
Nov 25 00:07:25 volumio kernel: [<7f28200c>] (smb_init [cifs]) from [<7f287378>] (CIFSSMBQPathInfo+0x54/0x270 [cifs])
Nov 25 00:07:25 volumio kernel: [<7f287378>] (CIFSSMBQPathInfo [cifs]) from [<7f2a7128>] (cifs_query_path_info+0x60/0x144 [cifs])
Nov 25 00:07:25 volumio kernel: [<7f2a7128>] (cifs_query_path_info [cifs]) from [<7f29a5bc>] (cifs_get_inode_info+0x2cc/0x4c4 [cifs])
Nov 25 00:07:25 volumio kernel: [<7f29a5bc>] (cifs_get_inode_info [cifs]) from [<7f29bc70>] (cifs_revalidate_dentry_attr+0xe4/0xf4 [cifs])
Nov 25 00:07:25 volumio kernel: [<7f29bc70>] (cifs_revalidate_dentry_attr [cifs]) from [<7f29bd40>] (cifs_getattr+0x60/0x12c [cifs])
Nov 25 00:07:25 volumio kernel: [<7f29bd40>] (cifs_getattr [cifs]) from [<8015bdec>] (vfs_getattr_nosec+0x3c/0x50)
Nov 25 00:07:25 volumio kernel: [<8015bdec>] (vfs_getattr_nosec) from [<8015bee4>] (vfs_fstatat+0x70/0xb0)
Nov 25 00:07:25 volumio kernel: [<8015bee4>] (vfs_fstatat) from [<8015bf78>] (vfs_lstat+0x28/0x2c)
Nov 25 00:07:25 volumio kernel: [<8015bf78>] (vfs_lstat) from [<8015c6f8>] (SyS_lstat64+0x24/0x40)
Nov 25 00:07:25 volumio kernel: [<8015c6f8>] (SyS_lstat64) from [<8000fb40>] (ret_fast_syscall+0x0/0x1c)
Nov 25 00:07:25 volumio kernel: INFO: task sftp-server:1280 blocked for more than 120 seconds.
Nov 25 00:07:25 volumio kernel:       Not tainted 4.4.9-v7+ #884
Nov 25 00:07:25 volumio kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Nov 25 00:07:25 volumio kernel: sftp-server     D 805b3864     0  1280      1 0x00000000
Nov 25 00:07:25 volumio kernel: [<805b3864>] (__schedule) from [<805b3ddc>] (schedule+0x50/0xa8)
Nov 25 00:07:25 volumio kernel: [<805b3ddc>] (schedule) from [<805b41b4>] (schedule_preempt_disabled+0x18/0x1c)
Nov 25 00:07:25 volumio kernel: [<805b41b4>] (schedule_preempt_disabled) from [<805b5874>] (__mutex_lock_slowpath+0xac/0x164)
Nov 25 00:07:25 volumio kernel: [<805b5874>] (__mutex_lock_slowpath) from [<805b598c>] (mutex_lock+0x60/0x64)
Nov 25 00:07:25 volumio kernel: [<805b598c>] (mutex_lock) from [<7f281cd0>] (cifs_reconnect_tcon+0x84/0x320 [cifs])
Nov 25 00:07:25 volumio kernel: [<7f281cd0>] (cifs_reconnect_tcon [cifs]) from [<7f28200c>] (smb_init+0x30/0x78 [cifs])
Nov 25 00:07:25 volumio kernel: [<7f28200c>] (smb_init [cifs]) from [<7f287378>] (CIFSSMBQPathInfo+0x54/0x270 [cifs])
Nov 25 00:07:25 volumio kernel: [<7f287378>] (CIFSSMBQPathInfo [cifs]) from [<7f2a7128>] (cifs_query_path_info+0x60/0x144 [cifs])
Nov 25 00:07:25 volumio kernel: [<7f2a7128>] (cifs_query_path_info [cifs]) from [<7f29a5bc>] (cifs_get_inode_info+0x2cc/0x4c4 [cifs])
Nov 25 00:07:25 volumio kernel: [<7f29a5bc>] (cifs_get_inode_info [cifs]) from [<7f29bc70>] (cifs_revalidate_dentry_attr+0xe4/0xf4 [cifs])
Nov 25 00:07:25 volumio kernel: [<7f29bc70>] (cifs_revalidate_dentry_attr [cifs]) from [<7f29bd40>] (cifs_getattr+0x60/0x12c [cifs])
Nov 25 00:07:26 volumio kernel: [<7f29bd40>] (cifs_getattr [cifs]) from [<8015bdec>] (vfs_getattr_nosec+0x3c/0x50)
Nov 25 00:07:26 volumio kernel: [<8015bdec>] (vfs_getattr_nosec) from [<8015bee4>] (vfs_fstatat+0x70/0xb0)
Nov 25 00:07:26 volumio kernel: [<8015bee4>] (vfs_fstatat) from [<8015bf78>] (vfs_lstat+0x28/0x2c)
Nov 25 00:07:26 volumio kernel: [<8015bf78>] (vfs_lstat) from [<8015c6f8>] (SyS_lstat64+0x24/0x40)
Nov 25 00:07:26 volumio kernel: [<8015c6f8>] (SyS_lstat64) from [<8000fb40>] (ret_fast_syscall+0x0/0x1c)
Nov 25 00:07:26 volumio kernel: INFO: task sftp-server:1294 blocked for more than 120 seconds.
Nov 25 00:07:26 volumio kernel:       Not tainted 4.4.9-v7+ #884
Nov 25 00:07:26 volumio kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Nov 25 00:07:26 volumio kernel: sftp-server     D 805b3864     0  1294      1 0x00000000
Nov 25 00:07:26 volumio kernel: [<805b3864>] (__schedule) from [<805b3ddc>] (schedule+0x50/0xa8)
Nov 25 00:07:26 volumio kernel: [<805b3ddc>] (schedule) from [<805b41b4>] (schedule_preempt_disabled+0x18/0x1c)
Nov 25 00:07:26 volumio kernel: [<805b41b4>] (schedule_preempt_disabled) from [<805b5874>] (__mutex_lock_slowpath+0xac/0x164)
Nov 25 00:07:26 volumio kernel: [<805b5874>] (__mutex_lock_slowpath) from [<805b598c>] (mutex_lock+0x60/0x64)
[... cut ...]
Nov 25 00:07:26 volumio kernel: [<8015bdec>] (vfs_getattr_nosec) from [<8015bee4>] (vfs_fstatat+0x70/0xb0)
Nov 25 00:07:26 volumio kernel: [<8015bee4>] (vfs_fstatat) from [<8015bf78>] (vfs_lstat+0x28/0x2c)
Nov 25 00:07:26 volumio kernel: [<8015bf78>] (vfs_lstat) from [<8015c6f8>] (SyS_lstat64+0x24/0x40)
Nov 25 00:07:26 volumio kernel: [<8015c6f8>] (SyS_lstat64) from [<8000fb40>] (ret_fast_syscall+0x0/0x1c)
....

And that's not even half. It's not having a great time after that; will reflash, as I say, and start afresh.

vincentgalore, 2016-11-25:
Ah. The system / web UI has just come back online. Will put that down to network issues, and update this thread with a new log shortly.

vincentgalore, 2016-11-25:
Well. I feel a little bit silly...

Yes, as the plugin pulls the shared network drive / folder and displays it next to its radio button, upon selection it copies the directory name into the 'Alias' and - more importantly - the 'Path' field with the /t whitespace. 

Having now checked that, and removed it, the drive / folder mounts perfectly and has infact restored all available network mounts via the plugin/system_controller-networkfs web UI, so they can be edited and removed.

I almost want to apologise for wasting your time...

xipmix, 2016-11-25:
No problem at all, this is a real usability issue.
Can you clarify - are you saying the Path field is getting
auto-filled from some other field, and there was a \t in that field?
Which field did you edit when you "checked that, and removed it: ?

vincentgalore, 2016-11-29:
Sorry for not picking that up. Yes, I attached an image; the 'Alias' and 'Path' fields populate upon selection of the network drive radio button after scan and the hostname dropdown menu. Perhaps including some kind of form validation that removes /t and all whitespace. What is odd is that the directory itself, as it is named on my hard drive, does not include a space. May have to play with that...

[image]