volumio/Volumio2 #315: [rc1] NAS mounts not showing under Settings

atcapollo, 2016-03-31:
NAS mounts (cifs) are not showing under Settings > My Music > Network Drives. Despite being successfully mounted. Therefore, they cannot be edited or deleted through the GUI.

However, the NAS folders are appear under Browse > Music Library > NAS > <share name>

xxxbugxxxx, 2016-03-31:
same problem on a rpi 2

apinner, 2016-04-04:
@xxxbugxxxx is this the same issue you was reporting in https://github.com/volumio/Volumio2/issues/313

Weasy666, 2016-04-12:
I'm having the same porblem on rpi 1b.

xxxbugxxxx, 2016-04-12:
Sorry for the late response. Same problem on RC1 on rp2. No problem for volumio 1.55 though.

xxxbugxxxx, 2016-04-12:
I have flashed back to 1.55 and did not keep screenshot.
Basically I setup a smb share. Mount the share using voluminous 2's web interface under library. The prompts showed successfully mounting.  However the setting does. Or get saved. Nothing stayed in the library page.

when I go to the tab to find the bad share , it does not show up either.  

Hope this helps

jag75, 2016-04-16:
Same issue with Raspberry pi3
Reproduce path;
1. Add network share
- Successfully added share
- Scanning starts, all seems ok

2. After reboot no share / library anymore exists
3. Add new share
4. Previous share will appear + new share
5. Reboot....no shares......

jag75, 2016-04-16:
Yes this was the problem.
I per accidentally added a share without using advanced options to add credentials in first place.
This will cause the issues, shares not visible at all.

You wil need to do this in order to see previous added shares but you will nto be able to delete them in the Network Configurations
. Add network share
*Successfully added share
*Scanning starts, all seems ok
2. After reboot no share / library anymore exists
3. Add new share
4. Previous share will appear + new share
5. Reboot....no shares......

xxxbugxxxx, 2016-04-16:
@jag75 LoL, yep that's the issue.

jag75, 2016-04-19:
LIbrary scanning is very instable. 
Only if I let it run for many hours it completes sometimes.
After a reboot the library is empty again and share is gone.
Volumio 1.5 works flawless, no issues at all with same library.

xxxbugxxxx, 2016-05-13:
Still open problem?

apinner, 2016-06-27:
Can anyone confirm if this is still an issue in RC2?

volumio, 2016-06-27:
Not an issue anymore

closed 2016-06-27 by volumio

xxxbugxxxx, 2016-06-30:
still an issue

jackhandyside, 2016-11-23:
Hi all. I'm afraid we are having the exact same problem with mounting and seeing network drives via the web UI / system_controller-networkfs tab. The issue is consistent, and remains even after multiple reboots, power cycling and factory reset of the unit. Attempting a fresh image now, and troubleshooting the network here.

What could be causing this and how do we resolve it?