volumio/Volumio2 #316: [rc1] Volume steps under logarithmic volume control feel too large

atcapollo, 2016-03-31:
The button volume steps of 10 under natural logarithmic volume control do not feel to be granular enough. Using i2s Iqaudio Pidac the 10 step in volume feel too big.

Feature request:

Can the volume steppings be added to settings to be user defined. The stepping granularity may need to change depending on the dac, hw control and whether linear or log controls are in use.

apinner, 2016-04-04:
@kurtommy this one is for you i think. 

I have to agree, 10 is quite a big step. Maybe 5?

kurtommy, 2016-04-04:
A configurations setting could be a good idea!

macmpi, 2016-04-04:
clicking on the wheel changes the green part, but not the actual volume, nor value. Then if you click on speakers, you may en-up with wierd ups/downs...

volumio, 2016-04-04:
We've already fixed that... You'll find it in next update

closed 2016-04-04 by volumio