volumio/Volumio2 #321: [RC1]Scanning of Music Library crashes

hmveldman, 2016-04-01:
When scanning music library on NAS (Cifs and NFS both) the scanning stops at a certain point leaving library empty.
The formerly indexed files stops playing immediate.
In the log i get the next error:
`info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
MPD error: Error: This socket has been ended by the other party
MPD error: Error: This socket has been ended by the other party
MPD error: Error: This socket has been ended by the other party
info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
MPD error: Error: This socket has been ended by the other party
MPD error: Error: This socket has been ended by the other party
MPD error: Error: This socket has been ended by the other party
info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Missing error handler on `socket`.
AssertionError: false == true
    at MpdClient.sendCommand (/volumio/node_modules/mpd/index.js:86:10)
    at ControllerMpd.getMyCollectionStats (/volumio/app/plugins/music_services/mpd/index.js:1231:17)
    at CoreCommandRouter.executeOnPlugin (/volumio/app/index.js:381:29)
    at Socket.<anonymous> (/volumio/app/plugins/user_interfaces/websocket/index.js:981:43)
    at emitTwo (events.js:100:13)
    at Socket.emit (events.js:185:7)
    at Socket.onevent (/volumio/node_modules/socket.io/lib/socket.js:335:8)
    at Socket.onpacket (/volumio/node_modules/socket.io/lib/socket.js:295:12)
    at Client.ondecoded (/volumio/node_modules/socket.io/lib/client.js:193:14)
    at Decoder.Emitter.emit (/volumio/node_modules/component-emitter/index.js:134:20)
info: `

volumio, 2016-04-01:
Please paste the log of journalctl into pastebin

apinner, 2016-04-04:
@hmveldman are you able to the log using http://pastebin.com/

hmveldman, 2016-04-04:
Do you want the log i already pasted in the post?
Or another one?

apinner, 2016-04-04:
The log of journalctl that includes the error. It makes it easier to follow and read in pastebin. Thank you.

apinner, 2016-04-11:
No response, reopen if necessary.

closed 2016-04-11 by apinner