volumio/Volumio2 #322: [RC1] Navigating music library does not work in Safari 9.1

Turbofoehn, 2016-04-01:
Hi everyone,

In the 'Browse' tab, clicking on a list item such as 'Music library' has no effect. It works in Chrome.

I am using 10.9.5.

apinner, 2016-04-03:
I don't quite understand the issue, what storage are you using?

apinner, 2016-04-04:
Anyone else with a Mac that can confirm this?

It's really a UI issue.

arnodeclercq, 2016-04-04:
I have some issues with iPad. 
Can't click on these links. They get highlighted but tapping, double tapping, holding, ... does not do anything. So it's probably Safari that's acting weird.
[image]

apinner, 2016-04-04:
@kurtommy you are a Mac user aren't you. Have you had any issues with Safari?

sbkivitz, 2016-04-04:
I can confirm this behavior on an iPhone with Safari.  I had started to debug it with the developer console and found that it was an issue with one of the supporting libraries (jsearch?), but for the life of me - I can't remember which one.  I will try to post more this evening when I'm home with my Pi3 / RC1.

volumio, 2016-04-04:
Yes, please. The more we know, easier we'll solve this

hicaoc, 2016-04-05:
I use safari  9.1 (11601.5.17.1) on osx 10.11.4  is ok . 

[image]

skikirkwood, 2016-04-05:
I am also using Safari 9.1 on OS X 10.11.4 for RC1 and everything is working fine.

On both my iPhone 6s and iPad 2, running the latest version of iOS, clicking through the UI is fine until there is a list of songs (or webradios) to choose.  Clicking on the row does nothing, you have to click on the icon of the 3 bars and bring up a pop-up menu to perform an operation on the row (which does work).

apinner, 2016-04-05:
Yeah just single clicking/pressing on the row that is normal behaviour, you have to double click in order to play or open the hamburger menu.

skikirkwood, 2016-04-05:
Ah, my bad.  Well, I can confirm that double-clicking in Safari on iOS with both an iPhone and iPad does nothing in RC1 for lists of songs or webradios.  But I just brought up Chrome on my iPhone and double clicking a list of songs or webradios also does nothing.  So it seems the problem is not Safari, it's iOS.

Double clicking in the Safari browser on my Mac works fine in the RC1 UI.

skikirkwood, 2016-04-05:
BTW, double clicking on songs and webradios does not work in Volumio 1.55 on either Safari or Chrome on an iPhone.  That's why I didn't realize I needed to double-click on the item in RC1, double-clicking has never worked for me on iOS.

apinner, 2016-04-05:
Double tap in chrome on android works, its starts the song playing.

skikirkwood, 2016-04-05:
I just got out my Amazon Kindle HDx 7, which runs a heavily customized fork of Android.  In a side-loaded Chrome browser, double-clicking doesn't work, but double-clicking in the default Silk browser on the device does work!

So it really looks to be primarily an iOS issue.  The UI for both 1.55 and 2 is certainly a lot smoother being able to double-click a track versus bringing up the hamburger menu.

apinner, 2016-04-05:
Weird.. I use the same tablet running cyanogenmod and chrome and double tapping works as expected. Also works on my Moto G running pretty much stock android.

skikirkwood, 2016-04-05:
I do have an old version of Chrome on my Kindle, let me see if I can find a recent APK to see if that helps.

skikirkwood, 2016-04-05:
Just installed the latest Chrome browser on my Kindle and double-clicking in RC1 does work.  Ok, looks like it's purely an iOS issue!  Good to have narrowed it down.

skikirkwood, 2016-04-05:
Apparently double-clicking in iOS in Safari is reserved for zooming in on the screen, so that's why it's not triggering any events.  Here's an old article that describes a workaround with JQuery:

http://appcropolis.com/blog/howto/implementing-doubletap-on-iphones-and-ipads/

apinner, 2016-04-06:
Was you aware of the above @kurtommy, no double tap on ios? :/

sbkivitz, 2016-04-06:
Hi, sorry about the delay--

Here is a screenshot of a Safari debugger connected to my iPhone 6 / iOS 8.3
[image]

This is the same error no matter which menu entry I try to click on (Favourites, Playlists, Music Library, Webradio).

How else can I help?  Seems @skikirkwood is onto something - but why would it work in some iPads / iPhones and not others?

sbkivitz, 2016-04-06:
Drilling down a bit further in the debugger tab...
[image]

This is jQuery JavaScript Library v2.1.4 - as indicated by the text header on this script

kurtommy, 2016-04-06:
@sbkivitz you spot the problem, is the lack of performance API. I'll fix that easily :)

skikirkwood, 2016-04-06:
That's strange.  I'm running iOS 9.3.1 on both my iPhone 6s and iPad 2, just hooked up the Safari debugger on my iMac to both, and got no errors at all on the Volumio 2 RC1 pages on either device.

That said, I think this is totally orthogonal to the issue that my understanding now is that Safari in iOS captures double-click events for screen zooming, so it will always be a NOOP for Volumio.  

I am also wondering why not just choose a single click event for selecting a song or webradio to play.  That's the default UI for most music applications on mobile devices (e.g. Spotify), so it's what people are used to.

closed 2016-05-29 by volumio