volumio/Volumio2 #325: [RC1] Buffer always plays out when restarting webstream after pauze

arnodeclercq, 2016-04-04:
When a webstream is paused and restarted later, by use of pause/play button, the webstream (buffer) plays a couple of seconds before stopping and restarting.
Maybe the buffer could be reset when someone pauses a webstream so when you continue the stream it's immediately correct.

apinner, 2016-04-09:
Listening to the radio this morning, I can confirm that this is pretty annoying!

volumio, 2016-04-15:
Any idea on how to solve this?

apinner, 2016-04-15:
Presumably the buffer that gets played is the mpd audio_buffer_size? 

I suppose the nicest solution is when a radio stream is playing the pause icon is replaced with a stop icon and the audio buffer is set to 0kb temporarily?

volumio, 2016-04-15:
Yeah, like the idea...

volumio, 2016-04-15:
Done it here: https://github.com/volumio/Volumio2/commit/d5b2492dc27192c85fe7b11540dfa4f5c163f400

Thanks for reporting

closed 2016-04-15 by volumio

apinner, 2016-04-15:
@volumio i was about to commit to the UI so a stop icon is displayed rather then pause when playing webradio.. Do you still want me to commit?

apinner, 2016-04-15:
https://github.com/volumio/Volumio2-UI/pull/90/commits/2fd75c7d57dabc669f5a9ec7001011d81948d87b

THEJAJO, 2016-04-21:
Something is wrong as I don't see Stop button, when I have only webradios on my queue and hit Pause.

apinner, 2016-04-21:
It will be in the next release