volumio/Volumio2 #329: Startup sound: please allow playlist !

macmpi, 2016-04-04:
Hi,

Alarm function is neat, but it requires the pi to stay on.
In some circumstances one may want to have the pi powered-on externally, and have Volumio play a defined playlist at startup.
Would be nice to allow selecting a playlist in lieu of startup chime
Thanks for consideration!

apinner, 2016-04-05:
So what you are describing is an auto start or auto play option after start-up?

macmpi, 2016-04-05:
Hi,
Thanks for looking into that.
I would say it's rather an auto-play after start-up.
Much like today, there is a default tune played after start (which can be disabled in settings).
Then the request is: allow to set a Playlist in lieu of that default tune.
Of course either tune or playlist playback on start-up could still be disabled.

Is it clearer?

apinner, 2016-04-05:
I would say they are separate features. What you want is the additional functionality of being able to set a playlist to auto play after start up not to simply replace the default start up sound.

macmpi, 2016-04-05:
your call.
In most cases I think I would then disable the default startup sound then: I don't think I'd want to hear both.
If in the end, you intend to let the possibility to have both, let's make sure the default startuo sound comes first: was not necessarily the case with Volumio 1.55, when adding a playlist in [rc.local](https://volumio.org/forum/autoplay-startup-t3420.html)

volumio, 2016-04-15:
I plan to do it, but it requires quite some work on the UI... As we need a selector that shows in real time changes to the fields ( select to select beetween startup sound & playlist playback, if playlist playback we need to show an additional field with the playlist list..)

apinner, 2016-04-15:
Would that just be a simple ng-if/show on the playlist select?

volumio, 2016-04-15:
Yep, something like the one we already have with visibleif

https://github.com/volumio/Volumio2/blob/d68780e06b16d8a96369a9fffd73295f9d8f6f23/app/plugins/system_controllers/network/UIConfig.json

but for selects...

macmpi, 2016-04-15:
or you may have a "master" switch for startup sound (on/off), with below 2 mutually exclusive tickable buttons, one for legacy tune, one (with field select) for playlist.

Worst case you could hard-code the startup playlist by a given name ("Startup"). Should user create or fill such playlist, then it would play at startup...