volumio/Volumio2 #332: Volumio 2, lack of direct transfer

Mairefe, 2016-04-05:
I could have accessed volumios(1.55) database from Osx and add songs / delete whichever I want. Sadly I couldn't do the same thing with this Volumio 2. Not only being "unseen" by osx, folders look like they don't contain music. Am I missing something here?

apinner, 2016-04-05:
Do you mean the smb Music share on 1.55?

Mairefe, 2016-04-06:
I don't know by that name. It was a simple dragn drop to add music files wirelessly. But now I can't after vol2.

Can I?

apinner, 2016-04-06:
I think 1.55 had a share that was a sym link to your NAS mounts. I believe that is due to return in v2 but hasn't been implemented yet. @volumio can you confirm?

Mairefe, 2016-04-06:
Guys dragn drop was a huge convenience. I have placed it beheath a mess of cables and its playing the key role in my hifi setup :)

And yeah I can remove the external harddisk everytime I want to add or remove. But it rescans the whole library which will take around 10mins..that is horror! :)

thomasdoerr, 2016-04-09:
Hi, I noticed the same and as far as I see, smb worked in 1.55 and in 2 it doesn't. It would be nice, if that would work again. Thanks!

macmpi, 2016-04-11:
Yes OP's request is directly linked to smb share, that conviently exposed 3 shares in Volumio 1.55:
NAS Music
RamPlay
USB Music
WebRadio

Hope this will come back.

apinner, 2016-04-12:
I am sure i have read that this is coming back, i can't find it now though. @volumio can you confirm?

volumio, 2016-07-04:
It was implemented in RC2 hotfix, with just one share: Internal Memory

Mairefe, 2016-07-04:
Oh okay. Thanks

Even though its good, its still quite unlike 1.55's transferring to usb music. (That is my external hdd in this case) A microsd card can't hold much.

But if "internal memory" is the usb music of 1.55, that is perfectly okay.

closed 2016-07-04 by apinner

volumio, 2016-07-04:
Sure we can add USB to shares

Mairefe, 2016-07-04:
Please notify us when you do!