volumio/Volumio2 #333: [RC1] Menu should close/hide when clicking outside of it

arnodeclercq, 2016-04-06:
Maybe this is mainly a personal preference but I think it would be better if the menu (that pops up when clicking the cogwheel) closes/hides when clicking outside of it.
Now you can navigate to another page while it stays open. Which, in my opinion, is not the best UX.

apinner, 2016-04-07:
As you say, that's a matter of preference. I don't mind it staying open.

I suppose the question is, whats the problem with it staying open or whats the benefit of it closing?

arnodeclercq, 2016-04-10:
It would be consistent.
If you click on an item in the menu it navigates to that page and closes the menu.
If you click outside of it and, for example, click on the browse link it navigates to that page but leaves the page open. Plus, it saves a click.
Also, on mobile it would be easier if the button to close the menu is located where the cogwheel is that opens it. Like how it's done on desktop.

apinner, 2016-04-12:
@kurtommy any thoughts?

kurtommy, 2016-04-12:
We'll do those changes in the next sprint :)

apinner, 2016-05-17:
Has been resolved and will be in the next release.

closed 2016-05-17 by apinner