volumio/Volumio2 #334: Import custom radio stations in .pls and .m3u format to Volumio 2

heis2201, 2016-04-06:
Often, people have their favorite audio streams setup as .pls or .m3u files which is well supported in Volumio 1.5. I am hoping for a function in Volumio 2 enabling the import of .m3u and .pls audio stream files. Either adding them to the existing web radios, handling them as playlists, or just playing them as if they were audio files...

apinner, 2016-04-12:
Related to https://github.com/volumio/Volumio2/issues/332

heis2201, 2016-04-12:
So, that means that I can simply copy my "old" .pls and .m3u files to somewhere (/var/lib/mpd/music/webradio ?)and they will appear in the player?

volumio, 2016-04-12:
Yes, I guess this is the idea... And it's a good one so I'll implement it...

heis2201, 2016-04-12:
Great! Thanks!

apinner, 2016-04-12:
@volumio what happened to the idea of the volumio selection of stations?

volumio, 2016-04-12:
Still on todo list... I've also created an API server based on parse... Do you think we could use it?

apinner, 2016-04-12:
I'm not familiar with Parse. Isn't it shutting down?

volumio, 2016-04-12:
They open sourced it... 
https://github.com/ParsePlatform/parse-server

Any alternatives? Or API server (ideally json based) you're familiar with?

aidanreilly, 2016-08-24:
would be really nice to see this in.

Cheddoleum, 2016-09-28:
Interestingly it does read .m3u files and include them after a scan of a music folder; and you do get the list of queue/play options -- it just won't actually play the URLs in the file.

heis2201, 2016-10-05:
Any update or recipe on how to do this best?

buzink, 2016-12-26:
Better support for custom radio streams would be nice. I think it would be most transparant to simply have a place somewhere in the browse tab where you can paste any url. Volumio could convert streams to the json playlist format and save them. Later you could even add support through plugins for youtube, soundcloud and podcast urls. All through the same add-a-url interface.

heis2201, 2022-01-31:
Dear all,
this issue is nearing its 6th birthday soon. Should we close it considering that there is manual way to achieve an import of m3u radio playlists (https://github.com/suuuehgi/volumio2m3u)?