volumio/Volumio2 #337: wireless IP address changes from fixed to a DHCP address

clevenp, 2016-04-07:
Using Volumio V2 build volumio-0.861-2016-03-25-pi.img
I set up a fixed wireless address using the file "interfaces" in the directory /etc/network
set up is as follows

auto wlan0
iface wlan0 inet static
address 192.168.20.64
netmask 255.255.255.0
gateway 192.168.20.1
wpa-ssid "T*********"
wpa-psk "++++++++++++++++++"

after booting , ifconfig shows the correct address
the address is a free address and no other device will request that IP address.

somewhere during the day, I cannot connect anymore to the UI and when i check the router, the device connected with the following address 
192.168.20.117 (a free DHCP address)
the address can be different if for example I had more devices connected to the router... 
It obvious requests a DHCP address and it gets the next free one

the console does not show any error except " exceeded CRDA call max attempts. Not calling CRDA"

I have noticed this change of IP address very frequently. Rebooting the device by using the UI command does not solve the problem and a hard shutdown - PowerUP is required to get the fixed IP address back.

Anjerlaan, 2016-04-07:
I have also a networking problem with V2 RC1 on a RPi B with 4GB SD:
I set a Static IP address in the WebUI (WIN10 with FireFox)
Save (Message shows that it saved the configuration), however after refreshing the network config is back to Dynamic. Also after reboot. (tried both wired LAN and WLAN)

volumio, 2016-04-10:
I'll work on it this week

volumio, 2016-04-10:
Should be fixed with : 
https://github.com/volumio/Volumio2/commit/d68780e06b16d8a96369a9fffd73295f9d8f6f23
https://github.com/volumio/Volumio2/commit/f6ce00a509527f12bd0f4771d9e296c271c795dc

New build will come soon...

closed 2016-04-11 by apinner

Anjerlaan, 2016-04-11:
Thanx i'll check it asap.
Is it now also possible tot set a static wireless IP address?

Greetzzz,

Gerben

Verzonden door BlueMail

Anjerlaan, 2016-04-11:
Oh sorry i see the answer in your changed code...