volumio/Volumio2 #338: Probllem with I2S DAC

jncraig, 2016-04-07:
I have several problem that happen when I turn I2S DAC on and select HiFiBerry from the resulting dropdown.

First, I need to reboot.  When I do, the Pi doesn't come back working.  If I power down and start again it does come back.

When it's back, the dropdown to select the Output Device has at least some display issues.  

Further, it won't play through the DAC.

If I turn I3S DAC off, reboot (with a power cycle), and then select the audio jack as output, I can't play most files.  I get "Error can't decode <filename>"  It will, though play webradio.

apinner, 2016-04-07:
> When it's back, the dropdown to select the Output Device has at least some display issues.
Like?

apinner, 2016-04-18:
No response so closing, reopen if necessary

closed 2016-04-18 by apinner