volumio/Volumio2 #339: WiFi Issue

jncraig, 2016-04-07:
I can now (RC2) select WiFi and it seems to be much better than previous betas.  But, if I unplug the ethernet cable, I can't connect using WiFi alone.

apinner, 2016-04-07:
How are you trying to connect? volumio.local or the WiFi IP address?

jncraig, 2016-04-09:
If I connect to WiFi, reboot and disconnect the ethernet cable before rebooting starts, then it comes up with the working WiFi connection.

apinner, 2016-04-11:
Are you saying its resolved?

jncraig, 2016-04-11:
I thought that it was.  It isn't.  I turned off the Pi and restarted it.  When it came up, it had lost the WiFi connection and did not reconnect.

apinner, 2016-04-11:
Still not sure what your actual problem is, but it may have been resolved here; https://github.com/volumio/Volumio2/issues/337#issuecomment-207998248

jncraig, 2016-04-11:
Thanks, I'll await the next RC!

closed 2016-04-11 by jncraig