volumio/Volumio2 #341: NFS does not seem to work anymore

gkkpch, 2016-04-09:
When I try to mount an nfs share, Volumio displays succesful mounting popups, but in fact it does not mount at all and displays the mount as false in the "My Music" tab.
This is what I find in volumio.log
`Error Mounting Share nfs: exec error Error: Command failed: /bin/sh -c /usr/bin/sudo /bin/mount -t nfs 192.168.123.250:/volumio-test /mnt/NAS/nfs
mount.nfs: access denied by server while mounting 192.168.123.250:/volumio-test`

This is how I have defined my share on the NFS Server (/etc/exports):
`/volume1/volumio-test	*(ro,async,no_wdelay,all_squash,insecure_locks,sec=sys,anonuid=1024,anongid=100)`
This has worked perfectly in earlier version sof Volumio 2 when we were investigating the utf-8 issue.

The alias parameter I used is "music-nfs".
The IP address I used you can see above.
The path is "/volumio-test"
Advanced options: FS type "nfs", user "guest", no options

When I use another user, e.g. "admin" with the corresponding password, and/or additional options, the error in the log is the same. None of it is reflected in the mount command.

There is another strange thing.
When I already have a cifs share defined (alias "MediaNAS-1", same ip, path "music") and try to add the  nfs mount, then Volumio complains about something already being mounted on "/nas/NAS/MediaNas-1", see volumio.log:
`No correspondence found in configuration for share nfs on IP 192.168.123.250
2016-04-09T09:37:57.438Z - info: [1460194677438] Error Mounting Share MediaNas-1: Something is already mounted on /mnt/NAS/MediaNas-1
2016-04-09T09:37:57.461Z - info: CoreCommandRouter::executeOnPlugin: networkfs , listShares
2016-04-09T09:37:57.685Z - info: [1460194677685] Error Mounting Share nfs: exec error Error: Command failed: /bin/sh -c /usr/bin/sudo /bin/mount -t nfs 192.168.123.250:/volumio-test /mnt/NAS/nfs
mount.nfs: access denied by server while mounting 192.168.123.250:/volumio-test
`

gkkpch, 2016-04-09:
OK, some of it I take back :)
The path should have been "volume1/volumio-test", then it works
I will re-investigate adding an nfs share when a cifs is already present.

gkkpch, 2016-04-09:
OK, I re-opened the issue because the second thing (adding a second library) does cause problems.
I tested this with a clean setup.
First added an nfs share (this time with the correct settings) and let mpd scan the library.
After it was finished, I tried to add a cifs share and Volumio complains about "something is already mounted on...", see above.
Obviously it does a check before mounting, but uses the wrong path.

Another issue: when you delete a share, it is unmounted and removed from the configuration settings correctly. It would also be good if the alias directory would also be deleted from /mnt/NAS so everything is clean again.

volumio, 2016-04-10:
Probably the best way would be avoiding to use the library for mounting, and develop our own code...

gkkpch, 2016-04-10:
I'm going to have a closer look at this. Javascript is not impossible when I can put a few debug print lines in here and there. So let's see where the ship strands...

apinner, 2016-06-27:
Was this fixed?

volumio, 2016-06-27:
Yep!

closed 2016-06-27 by volumio