volumio/Volumio2 #343: [Volumio2] unable to connect from Volumio device to UI

escoand, 2016-04-11:
I wan't to use a touchscreen display at the volumio device for the UI.
But every connection to the UI is refused.

```
$ wget -O- http://volumio
converted 'http://volumio' (ANSI_X3.4-1968) -> 'http://volumio' (UTF-8)
--2016-04-11 07:05:11--  http://volumio/
Resolving volumio (volumio)... 127.0.0.1
Connecting to volumio (volumio)|127.0.0.1|:80... failed: Connection refused.

$ wget -O- http://volumio.local
converted 'http://volumio.local' (ANSI_X3.4-1968) -> 'http://volumio.local' (UTF-8)
--2016-04-11 07:05:52--  http://volumio.local/
Resolving volumio.local (volumio.local)... 192.168.8.8
Connecting to volumio.local (volumio.local)|192.168.8.8|:80... failed: Connection refused.
```

apinner, 2016-04-11:
try localhost:3000

escoand, 2016-04-11:
Now it is working, fine.
I created a little script for this: https://gist.github.com/escoand/11e712eac5780d6a386e4bcfe3d407d6

volumio, 2016-04-11:
Nice! Let me think on how to integrate in an easy way ...

escoand, 2016-04-11:
This would be really great.

I've updated the GIST again to use Ratpoison.
Otherwise Chromium will not run completely full screen.
I've tested also Iceweasel/Firefox and Midori but they have a bad touch support.

escoand, 2016-04-11:
BTW I use the display rotated because 800x480 needs scrolling on main screen.

volumio, 2016-04-11:
Got an idea...
Instead of using Chromium, we could pack everything into a crosswalk hybrid app, therefore customizing it even more..
https://crosswalk-project.org/

What do you think?

escoand, 2016-04-12:
I always prefer to be as lightweight as possible.
I don't want to have an app for every piece of software, not on Android, not on my desktop, and also not on the RPi.
If it's just for the same web interface and not more functionality this is just more work in my opinion.
Actually Chromium is for me already to bloat for showing and using the interface but it's the only usable touch browser.

escoand, 2016-04-12:
Additionally this unneeded overhead can get a showstopper for the project at all. I've used many promising open source projects but at some point they get too big and the leisure time too less. And so they died...
I would cite the Unix philosophy: do one thing and do it well.

apinner, 2016-04-12:
Can we close this one or is it ongoing?

escoand, 2016-04-12:
My problem is solved. Don't know if @volumio needs it as todo  entry or for further discussions

volumio, 2016-04-12:
Let's keep it open, as I would like to make a plugin for this... We're building the plugin system as we speak and this may be a good candidate

escoand, 2016-04-20:
There should be an option in that plugin for rotating the screen and the touch axis. I've changed the DAC setting and it overrides my rotate setting.
Or maybe better an option to set some custom settings and start commands.

volumio, 2016-04-20:
I cannot understand your first question..
And which options would you like to set at startup? Bash one?

escoand, 2016-04-20:
There are two things to do for rotating the screen.
1. display_rotate=1 in /boot/config.ini
2. execute this commands
xinput --set-prop "FT5406 memory based driver" "Evdev Axes Swap" 1
xinput --set-prop "FT5406 memory based driver" "Evdev Axis Inversion" 0 1
Maybe the two commands can be in Xorg conf.

volumio, 2016-04-20:
Yes we can definetely add those options in the Plugin for touchscreen. We're preparing the plugin backend this week, so this can be added without issue...

closed 2016-11-03 by escoand