volumio/Volumio2 #344: Allow WoL for NAS

macmpi, 2016-04-11:
Hi,

Would be nice if Volumio allowed an option to send Wake-on-Lan "magic" packet to NAS server to wake them-up.

Thanks!

apinner, 2016-04-11:
I would say that most NAS devices and Servers etc are on 24/7 anyway. That combined with the fact that most instances of volumio are probably also kept on 24/7 due to them being low power. And the large number of phone and tablet apps that offer one touch WOL, I'd say this probably has limited usefulness.

There is also an issue https://github.com/volumio/Volumio2/issues/204 where if the NAS is powered on after volumio is doesn't mount.

macmpi, 2016-04-12:
Well...  If all NAS makers implemented WoL I guess it's due to customers demand for it. In personal / soho environments people tend to be energy conscious these days :)
Kodi & al (which may run in similar context) provide such useful WoL feature.

Issue #204 is indeed an issue that should be resolved in itself.
Once this #344 is resolved, it will indeed greatly benefit from #204 resolution.

apinner, 2016-04-12:
WoL is one of those things that can be tricky and frustrating to get working. For a start WiFi support isn't great and lots of SoHo routers don't support it. It can also be difficult to diagnose when it doesn't work and who knows what kind of hardware someone might be trying to wake. 

I see many many forum posts and github issues from people not being able to get it working.

macmpi, 2016-04-13:
[This](https://github.com/mandark/xbmc-script.advanced.wol) may help or provide inspiration (scheduling for library scans, mounts, etc)

apinner, 2016-05-17:
Pretty sure that this could be implemented with a plugin.