volumio/Volumio2 #345: Shutdown does not turn-off the device

macmpi, 2016-04-11:
Shutdown command does not power-off the pi: green led stays on on my rpi0.
Onscreen last message (hdmi) is: reboot: System halted.

Expectation would be that shutdown really turns-off the device.
Thanks.

apinner, 2016-04-11:
What happens if you use the volumio shutdown command via SSH? 

`sudo /sbin/halt`

macmpi, 2016-04-12:
Hi.

> sudo /sbin/halt

 has the same effect that Volumio UI shutdown, and therefore does not power-off the devices.

> sudo shutdown -h now

does power-off the device.

Hope this helps in narrowing this down.

apinner, 2016-04-12:
The rpi0 has no power led only an activity?

The ACT led also seems to be inverted on the rpi0, so on = no activity and off = activity.

Also taken from the rpi forums...

> If you have Wheezy installed using commands sudo halt, sudo poweroff or sudo shutdown -h now the led will flash 10 times and then turn off and the Pi will shutdown/halt.

> If you have Jessie installed using command sudo halt the led will flash off a couple of times and then stay permanently on after it has halted. Using commands sudo poweroff or sudo shutdown -h now the led will flash 10 times and then turn off and the Pi will shutdown/halt.

Looks like its normal behaviour.

macmpi, 2016-04-12:
Yes only 1 led on rpi0
Volumio 2 is the only distrib I have that keeps that led on after "shutdown". I have other Jessy distribs that turn it off on shutdown.
Volumio 1.55 did turn it off for instance.

Why isn't "sudo shutdown -h now" the preferred shutdown method for UI invocation?
As I recall "halt" does not send ACPI hardware signal that commands complete shutdown.

Thanks.

apinner, 2016-04-12:
I can't answer why.. perhaps @volumio can?

volumio, 2016-04-12:
Fixed! 
https://github.com/volumio/Volumio2/commit/dfcc21121b5752066b140de9dfaecaa35a82b601

closed 2016-04-12 by volumio

firemeteorxx, 2016-05-18:
I'm not sure if changing halt to shutdown is a reliable fix.
Just to provide some more data point. (BTW, I'm on RC1 so couldn't comment for sure if the fix works or not.)

The reason of for LED not turned off, as in my case, is that the shutdown process does not successfully reach its final state. My setup is using a wireless only network. And it appears that the wireless network configure flow is troublesome.
Upon boot, the wireless network is configured too late, increased the wait time before the system starts to response.
Upon shutdown, the wireless network **sometimes** shutdown too early -- earlier than the NFS unmount. In such situation, the NFS umount will get stuck and the shutdown process could not proceed.
However, I mentioned **sometimes** here since it does not always happen. Even though it happens quite often, there is still some chance that it reaches complete shutdown.

volumio, 2016-05-18:
Any idea on how we could solve it?

firemeteorxx, 2016-05-20:
I don't have answer:-(
I'm really not an expert on systemd -- much more familiar with the old sysv style.

apinner, 2016-05-20:
Couldn't we just force an unmount before we issue the shutdown command as part of the shutdown process?

macmpi, 2016-10-13:
It seems UI shutdown (and restart) may take more time now to fully complete (dismounting NFS share is ended by watchdog) and eventually not fully complete (leds ON).

macmpi, 2016-11-18:
BTW, seems like the proper [systemd way](https://access.redhat.com/documentation/en-US/Red_Hat_Enterprise_Linux/7/html/System_Administrators_Guide/sect-Managing_Services_with_systemd-Power.html) to do it is:
```
sudo systemctl poweroff
sudo systemctl reboot
```
You may want to update [platformSpecific.js](https://github.com/volumio/Volumio2/blob/master/app/platformSpecific.js) accordingly for shutdown and reboot.

volumio, 2016-11-19:
According  to the docs, your systemctl commands are equivalent to the ones we already have:

"Although the shutdown command will call the systemctl utility to perform the shutdown, it has an advantage in that it also supports a time argument."

macmpi, 2016-11-19:
Yes. Was just a comment for sake of consistency, since you tend to rather implement the systemd-way, and in current lines, time argument is not used.
Your call ;)