volumio/Volumio2 #346: Issue with HDMI CEC forced

macmpi, 2016-04-11:
Hi,

I Noticed an annoying issue when using Volumio2 with a HomeCinema amplifier (audio out through HDMI).
Though audio functionality is all good, there is no way to select another HDMI input on the HomeCinema system while Volumio is running: this may be cumbersome in many cases (wath images, but keep Volumio audio, check what's going on on another input, etc...).

The issue seems to come with CEC assertion that is forced.
Adding the following in config.txt fixed the issue.

> hdmi_ignore_cec_init=1

Wouldn't it make sense this is by default in config.txt?
pitfall: it does not switch HDMI port at startup.

There might be other/better way to achieve that, to allow both:
- HDMI switch to correct HDMI port when Volumio is initially started
- but then do not force HDMI port after that, even if Volumio is till active.

macmpi, 2016-04-12:
FWIW, Kodi integrates [Pulse-Eight libCEC](http://libcec.pulse-eight.com/) with highly customizable options in System -> Settings -> System -> Input Devices -> Peripherals -> CEC adapter.
Maybe that same lib could help adding more flexibility in CEC handling within Volumio 2 (as hdmi_ignore_cec_init=1 may not be panacea)

volumio, 2016-04-16:
Fixed with https://github.com/volumio/Volumio2/commit/221cdcfe5bf14c360cd6d4c83c66c688c2e027af

closed 2016-04-16 by volumio