volumio/Volumio2 #347: [RC1]Scanning of Music Library crashes

hmveldman, 2016-04-11:
My Volumio 2 rc1 setup is crashing during scanning of the NAS, mounted as NFS.
Setup:
raspberry 2 with hifiberry DIGI + clone.
Wired network
Take a look at the log
[http://pastebin.com/5mVYcj7b](url)

apinner, 2016-04-15:
Out of interest do you have any embedded artwork in those m4a files?

hmveldman, 2016-04-15:
Yes i do, but when i put my sd card with Volumio 1.55 in my raspberry the library is scanned perfectly.
The album of Lykke Li, where the scanning crashes is shown in Cantata, with artwork embedded.

jag75, 2016-04-17:
I have this issue as well.
Raspberry Pi 2 & 3
Volumio 2 RC1

Embedded artwork in media files (WAV / FLAC)
Volumio 1.55 works flawless
Volumio 2 RC 1 stops indexing after track x

apinner, 2016-04-18:
Ok... just out of curiosity, what happens if you remove the embedded artwork? Does it still fall over when trying to scan?

hmveldman, 2016-04-18:
Ah, that did the trick!
I replaced the artwork of the specific album and now it is scanning the whole library again.
Probably the jpg was somehow corrupted.
Thank you for putting me in the right direction!

apinner, 2016-04-18:
I had a feeling it was going to be down to corrupted album art. 

I'm not sure why mpd would crash though. @volumio do you have any ideas?

closed 2016-04-18 by apinner

volumio, 2016-04-18:
Probably is ffmpeg... I'm building a new version of mpd which should fix this

jag75, 2016-04-19:
I still have issues while scanning the library. It takes many hours to scan 600 Albums.
I let it run overnight and all albums seems to be indexed now.
A reboot sometimes lead to unsuccessful startup and I have to power it down and give it a hard restart.
After reboot the library is empty again and the share is gone.
Any idea?

apinner, 2016-04-19:
Reboots can be unstable, i have had issues with my development Pi not coming back after a reboot. I am sure i have seen this is being addressed.

I would think the amount of time to scan the library will depend on a lot of things outside of the control of volumio. The speed of the network, the speed of the nas/server, the total size of the library, the number of individual tracks, what else the nas is doing, other network traffic etc. I haven't had any problems with the library disappearing after a reboot though.

jag75, 2016-04-19:
I compare library scan form Volumio 1.55 with Volumio 2 RC1
Volumio 1.55 works without any problem and takes less than 5 minutes.
Volumio 2 RC1 seems to have issues using same library and takes many hours and reboot issues.
Doesn't look good to me.

apinner, 2016-04-19:
Can you post a log like hmveldman did?

I suspect you may also have embedded album art that it does not like.

jag75, 2016-04-20:
I have removed the embedded Album art and no issues anymore.
I doubt the Album art is corrupted though, it works fine on other streamer. (S.n.s)

One question and maybe I should open a question but it seems there is no way to get Album art configured in MPD (mPoD or mPad) anymore. I used to configure virtual link in Volmumio 1.5 but this is not working in Volumio 2 since it doesn't seem to use php?

apinner, 2016-04-20:
Was it also .m4a?

jag75, 2016-04-20:
affirmative

JoelMattisson, 2016-05-23:
I have the same problem, running on Raspberry 3. New to this, how do i access the log file?

volumio, 2016-05-23:
you can see what's going on by doing (while scanning)

sudo journalctl -f