volumio/Volumio2 #349: Music Library cant scanning sacd-iso  file ?

hicaoc, 2016-04-12:
I upgraded to support sacdiso play mpd 
music library scanning documents less than sacdiso file ？

apinner, 2016-04-12:
I don't understand what it is you are trying to ask?

hicaoc, 2016-04-12:
There are a lot of music files sacd-iso format on my NAS, update the music library when other music formats , such as ape, flac, dsf, dff can find and automatically added to your music library , and iso sacd image format files are not scanned.

apinner, 2016-04-12:
Looks like there was a forum thread https://volumio.org/forum/sacd-iso-direct-support-t3252.html

From what i can tell sacd iso is not officially supported by mpd and unofficial support is sketchy at best.

hicaoc, 2016-04-12:
yes，i see，
I use this version （https://github.com/hicaoc/MPD.git） is support for sacdiso and native dsd you web link the forum Mr. xxxbugxxxx today with this version of MPD  。
I have upgraded the mpd on volumio2 support sacdiso, but does not support volumio2-webUI

volumio, 2016-04-12:
Love your mpd fork, it seems to have all the features I was trying to get while merging the SACD fork and DSD one...
It is the perfect candidate to become the Volumio mpd... On which version its based on?

hicaoc, 2016-04-12:
volumio 1.55 and 2, I have tested, 1.55 worked well, volumio2 some are not compatible, you need to be adjusted volumio2
My fork in the mpd 0.20 based on open has not yet officially released, bug may be more, but also requires a lot of testing. But it can be combined over the same time for development and testing

hicaoc, 2016-04-12:
volume2  linux kernel supports native dsd , but libasound not support, you need to re-download the new version of alsa-lib .

xxxbugxxxx, 2016-04-13:
Tested on volume 1.55 works, just need alsa to be 1.0.29 and later. if it can be mereged to stock volumio that would be great. 

BTW @hicaoc , DIF file still has the faster speed problem form your repo. DSF file works fine.

xxxbugxxxx, 2016-04-13:
@volumio RPI 2 has all the support needed for native DSD+sacd ISO altogether using hicaoc's repo. It would be the ultimate volumio. still glitches still need to work out (DFF file speed)

hicaoc, 2016-04-13:
@xxxbugxxxx 
i fixed it . you try

xxxbugxxxx, 2016-04-14:
Works perfectly now Dff,dsf,iso

xxxbugxxxx, 2016-04-14:
@ hiCan't DOP mode also works, I suspect the mainline MpD has his 4x bug, not  my Gustard x20 dac.maybe more people can try to confirm

hicaoc, 2016-04-14:
dop ok，but dsd2pcm speed slow，i will fix it

xxxbugxxxx, 2016-04-14:
With your latest fix, native dad works for both dfm and diff.

And if I set it to dip mode, also works. Dsd256 in dop mode fallbacks to native dsd mode. So everything works for me, no dsd2pcm involved.

xxxbugxxxx, 2016-04-14:
Is it because of dsd u32? I thought MPd implementation of dsd u 32 is combining four dsd u8

xxxbugxxxx, 2016-04-14:
@hicaoc i can reproduce your dsd2pcm problem in your latest repo

hicaoc, 2016-04-14:
i fixed it dsd2pcm

closed 2017-06-02 by volumio