volumio/Volumio2 #352: Backup of Music Library

Weasy666, 2016-04-12:
It would be great if there were an option to backup and restore the music library.
So that one doesn't have to scan thousands of songs again after reinstalling volumio.

That could also be helpfull for setting up different volumios which are accessing the same NAS.

apinner, 2016-04-12:
You can manually backup the mpd database.

Something even more useful would be the ability to export and download a config file from volumio that can be uploaded to a new instance that contains all settings. Much like most routers.

volumio, 2016-04-12:
Yes... Or maybe a cloud backup...

apinner, 2016-04-12:
Lets get the hosted interface working first :P 

Cloud backup would fit with that really nicely.

Weasy666, 2016-04-12:
Yes, something like a import/export-functionallity for the settings and/or the library is exactly what i meant. I wouldn't even go so far and doing a cloud backup. For cloud you normally would need an account...i don't know if everybody would be fine with that

volumio, 2016-04-15:
@apinner any idea for this?

I would make this way:
- Button on System, in a new section called system backup 
- On Press we open a modal, with checkboxes for 
*music database
*albumart cache
*my webradios
* favourites
* settings 
* playlists
*all

Then 2 buttons: backup and cancel and a select Backup method: download or USB 

When you hit backup the API would be:
doBackup:{items:all|playlists|settings etc...}

I do it on the backend and then expose it via express route, and we send the download link via
backupDone:{ message: message, downloadLink: htttp..... (only if we selected download method).
Now, all we need to do and show the message and if there is a download, start the download automatically.

Then we need the restore backup button

Modal, we can drag n drop the backup archive there, and we need to upload it to the BE. We also need to add a button to check USB backup.
the API will be:
backupRestored : {status: messages that will notify about progress}

What do you think? Want to do it with me?

apinner, 2016-04-15:
I would probably do this under System and add a backup/restore panel there.

What you have sounds like a good plan. Happy to help and work with you where I can!

volumio, 2016-04-15:
Great!

volumio, 2018-03-26:
Done with a plugin

closed 2018-03-26 by volumio

maxdd, 2021-10-03:
Is the plugin Backup & Restore Data the right plugin?