volumio/Volumio2 #353: dirble webradio list only first page

balbuze, 2016-04-12:
I come back to an old problem : when listing radio by country, only the first page from dirble is displayed. In a previous issues ( I don't refind it) we were talking about an infinite scrolling so that having the whole list returned by dirble

apinner, 2016-04-12:
This issue? https://github.com/volumio/Volumio2/issues/136

balbuze, 2016-04-12:
Yep it was the base

gkkpch, 2016-04-13:
yes, we did check this a bit, when you manually enter an url and supply the page number, it seems to  work sort off, but even then I think there was an error with page 0 and page 1.
Still, good to know it is now listed as an issue.

apinner, 2016-06-27:
Dirble replaced with shoutcast, no longer an issue

closed 2016-06-27 by apinner

balbuze, 2016-06-28:
But dirble still use when sorting radio by country it seems.

apinner, 2016-06-28:
Which doesn't seem right, @volumio is that just a hangover that needs to be corrected?