volumio/Volumio2 #355: When running on low volume the output becomes static noise

janpieterz, 2016-04-13:
A fresh install of the RC1 build on a Raspberry PI 3 resulted (over here) having a static noise being generated when the volume is low. This effect occurs both using airplay (volumio volume and streaming device volume) and webradio. The noise becomes louder than the  (soft) music.

apinner, 2016-04-13:
Using which DAC?

janpieterz, 2016-04-13:
No DAC, just the normal audio jack for now. I know it isn't that great, but this seems a but worse than just 'bad quality' audio.

apinner, 2016-04-13:
Unfortunately that's the hardware, not the software.

Quoted from another source: https://discourse.osmc.tv/t/noise-on-analog-audio-line/4820

> I hate to be the bearer of bad news but the analog output of the Pi is quite noisy unfortunately, and at low volume levels you will also hear significant distortion. Nothing really that can be done about that in software other than to keep the master volume at 100% and make use of volume amplification etc as described in that other thread.
> 
> The analog output is just a simple PWM (pulse width modulated) output directly from the SoC passed through a very rudimentary analog low pass filter. Effective resolution is approximately 11 or 12 bits (nowhere near 16 bit CD quality - some 30dB less SNR in fact) and there also seem to be problems with correct dithering not being applied before the samples are truncated and sent to the PWM system. (Which causes the harsh sounding distortion at low volumes - also known as quantization noise)
> 
> There was some talk a while back of replacing the PWM routines in the Pi's firmware with Sigma Delta modulation which in theory could make a big improvement to the analog output quality but I've heard no more about that happening. That's something that the OSMC and Kodi devs can't do anything about though, only the engineers working on the Pi's firmware could implement such functionality.
> 
> For use plugged into a small pair of computer speakers or similar you might find the analog output satisfactory, but for plugging into a proper stereo/home theatre system which doesn't support HDMI sound input you're much better off getting an HDMI audio splitter which will split the audio out of the HDMI stream to some analog outputs - the quality will be MUCH better than the built in analog output will ever be.

janpieterz, 2016-04-13:
Ah, thanks for the clarification. Will check the available ones out this weekend. Shame really, was just playing around to see how viable the combination is.

closed 2016-04-13 by janpieterz