volumio/Volumio2 #357: Upper case caracters in player name becomes lower case

TheRealMattiasJohansson, 2016-04-13:
If i rename the player name from Volumio to, for example, "Living room" and then saves the changes the player name becomes all lower case, i.e. "living room".

I'm running Volumio 2 RC1 0.861 on a Raspberry Pi 2.

apinner, 2016-04-13:
I would imagine its because the player name is also the hostname, and most linux distros, including debian on which volumio is based force the hostname to be lowercase.

TheRealMattiasJohansson, 2016-04-13:
Sounds reasonable, but `the one wonders why the default player name "Volumio" (with an upper case V) is allowed and working?

apinner, 2016-04-13:
Haha, i did say 'i would imagine' not that i am correct :P

XiozTzu, 2016-05-11:
I have noticed that it, for whatever reason, requires 2 reboots for name changes to propagate to Airplay.  Seems silly, I agree should just happen on the fly when the name is changed.  The Airplay name needs to be decoupled from the hostname of the box.  (This would also fix the lowercase name issue)

xipmix, 2016-11-04:
This issue is closely tied to #293.

xipmix, 2016-11-04:
The issue of 'Volumio' vs 'volumio' seems to be happening in ControllerSystem.prototype.getUIConfig()

which does this:
```
self.configManager.setUIConfigParam(uiconf,'sections[0].content[0].value',self.config.   get('playerName').capitalize());
```
ie it is forcibly capitalized.

closed 2017-06-02 by volumio

2oh1, 2017-10-12:
This is still an issue.

In Volumio, under System Settings, the player name is forcibly capitalized - which is fine.  In fact, I typed the name capitalized in the first place, but even if I try typing it in all lowercase, it gets capitalized when I click away from the System Settings page and return to it.  Type it lowercase.  Save.  Go to a different page, then return to System Settings.  It's now capitalized...

...But the player name is usually being broadcast in all lowercase.  I say "usually" because if I go into Volumio System Settings and save anything, the player name usually changes to being broadcast as capitalized, but a restart will usually (but not always!) switch it back to all lowercase.  And sometimes, saving settings in Volumio can confuse Airplay receivers which will think there are now two broadcasts (one capitalized and another lowercase, because they're seen as being two different names).

2oh1, 2017-10-12:
The plot thickens.  /etc/hostname shows my hostname as all lowercase.  Where did that come from?  I never set a hostname.  I only changed the player name in Volumio System Settings.

If I manually change the hostname to be Capitalized, and then I change system settings, the hostname file is overwritten (the text in that file gets changed to being lowercase again even though it's capitalized in system settings).

xipmix, 2017-10-12:
The other issues you linked will have some of those details.

2oh1, 2017-10-13:
I don't understand.  [You found and documented this very problem last year](url=https://github.com/volumio/Volumio2/issues/293#issuecomment-258594338).  Was it ever addressed?

There are three variables here.

Player Name, as seen in Volumio System Settings.  It gets forced to be capitalized.

Hostname, as found in /etc/hostname.  It keeps getting forced into lowercase.

Broadcast Name.  In theory, this is the player name, but the broadcast name sometimes gets changed to lowercase after save or reboot even though the player name is still shown as capitalized.

Is there a way to override this behavior and always force a broadcast name to be the same?  I'm using Applescript on a Mac to switch playback from the Mac to my Pi, but for code to work, the name needs to be consistent.  The name change is especially problematic because iTunes sometimes waits, expecting to stream to the capitalized Airplay, but Volumio has changed itself to be lowercase, which iTunes sees as if it's a new (second) Airplay device.

2oh1, 2017-10-13:
I don't understand.  [You found and documented this very problem last year](https://github.com/volumio/Volumio2/issues/293#issuecomment-258594338).  Was it ever addressed?

There are three variables here.

Player Name, as seen in Volumio System

xipmix, 2017-10-13:
I think there is but it will probably require some code changes. As you can see from the history there's not much interest. I've been messing around with some code to canonicalise the various names according to the constraints but haven't had time to propose some concrete changes. 

Here's my take on your points above, interested to hear your impression:
 * Player Name should be whatever the user sets it to, though there should be some feedback when setting it so the user doesn't inadvertently choose illegal characters or make it too long etc.

 * hostname - should be independent of player name. Ideally dhcp would set the hostname. If the user is able to set it (would be a separate 'advanced' setting) the default should be whatever ends up in /etc/hostname once the system has booted. If the user wants to override that goes in a config.json and that name gets used thereafter. Since it's basically a DNS name it should always be all lower case and conform to DNS norms.

 * broadcast name - this seems to be a mDNS name (aka bonjour). Possibly this should just be the same as Player Name with no modifications. That would seem to be the least confusing, particularly for iTunes which doesn't seem to be particularly smart. If the fact it is an mDNS name means there are restrictions on the string that can be used, those should probably be enforced when setting Player Name.

There's also the 'netbios name', used for windows file shares. It defaults to hostname but there may be a desire to override it. Player Name could be used but there are constraints on the string eg a 15 character limit.

volumio, 2017-10-13:
The current behaviour is due to the fact that the latest shairport-sync version introduced a regression in conjunction with Volumio's mdns advertisement protocol. Basically if Shairport sync's name and MDNS name are the same, the system will crash. 
I know its ugly, but it's the only way I've found to avoid crashing of the MDNS advertisement protocol (see volumio_discovery plugin)

2oh1, 2017-10-13:
> "I know its ugly, but it's the only way I've found to avoid crashing of the MDNS advertisement protocol"

My issue isn't that it's ugly.  My issue is that it crashes Volumio (or Shairport-Sync?) and that it confuses Airplay players streaming to Volumio due to name changes (sometimes it's capitalized.  Sometimes it's lowercase.  The name needs to always always always always always be the same.

Currently, it sometimes changes after a save or after a restart, which leaves a player on the other end streaming TO Volumio looking for an Airplay receiver that is no longer there (because it's looking for Capitalized_Name but Volumio changed it to lowercase_name.  But if I try to change the name to always be lowercase, Volumio System Settings forces the lowercase name to become capitalized.

> "(see volumio_discovery plugin)"

Got a link?  I'm searching but can't find it.  I see a [reference to it here](https://github.com/volumio/Volumio2/issues/1114), but it's crossed out.

Is there a way to force the broadcast name to always always always be the same?

It seems to me, the fix should be separate settings for hostname and player name, instructing users to name them differently.

2oh1, 2017-10-13:
re: xipmix

> Player Name should be whatever the user sets it to, though there should be some feedback when setting it so the user doesn't inadvertently choose illegal characters or make it too long etc.

Absolutely.  And if the Player Name needs to be different than the hostname, build in a check to make sure the user hasn't given them identical names ("Sorry.  You've assigned that name as the hostname.  Please choose something different.  Perhaps an alternate spelling?")

> hostname - should be independent of player name.

Absolutely.  Tying them together causes crashes.

> broadcast name - this seems to be a mDNS name (aka bonjour). Possibly this should just be the same as Player Name with no modifications. That would seem to be the least confusing

Either that, or create a setting in System Settings for a specific Airplay Name for Shairport-Sync?

> If the fact it is an mDNS name means there are restrictions on the string that can be used, those should probably be enforced when setting Player Name.

Certainly, that makes sense.

> There's also the 'netbios name', used for windows file shares. It defaults to hostname but there may be a desire to override it

Perhaps there's a need for a System Setting for that as well - or a conf-selectable override for situations where it causes problems.

2oh1, 2017-10-13:
Further adding to the story...

I decided to see what would happen if I started over from scratch, running the config Wizard in System Settings.  This time, when it said "Choose a unique name for your **device**" I gave it a new name: volumio.   Then, in System Settings, I named the **player** victor.  Volumio forced the Player Name to be capitalized.  That's fine.  Whatever.

When starting up a stream of Airplay to Victor, Volumio crashed.  When it restarted, I went into System Settings and - here's the important part - the Player Name had been changed to Volumio.

Two things:

1. I never named the player Volumio.

2.  I named the device volumio, but in all lowercase, not capitalized.

So, it would appear that the Player Name can be overriden by the Device Name.

DarknessFX, 2018-06-07:
I had a problem with the Volumio changing my player name FXVolum to Fxvolum after reboots, I fixed with this steps:
##
File /volumio/app/plugins/system_controller/system/index.js
Line 88
Change : 
  self.configManager.setUIConfigParam(uiconf,'sections[0].content[0].value',self.config.get('playerName').capitalize());
To : 
    self.configManager.setUIConfigParam(uiconf,'sections[0].content[0].value',self.config.get('playerName'));

##

Thanks.