volumio/Volumio2 #359: Volume doesn't get set properly (bounces between levels)

ShanDestromp, 2016-04-14:
Raspberry Pi B+; NFS Database (hard wired GigE network) using RC-1

` uname -a
Linux volumio 4.1.19+ #851 Sat Mar 5 22:26:57 GMT 2016 armv6l GNU/Linux
`

Changing the volume level from whatever it's at on boot will cause it to bounce between it's original level and whatever you attempted to change it to (EG 50/75).  This will additionally cause the interface to become laggy (when operated via touchscreen) or non responsive (eg mobile or another networked computer).  At first I thought it was just the Pi being under-powered but even stopping playback and letting it sit for 10-15 minutes the volume level will keep bouncing around.

apinner, 2016-04-14:
Which browser are you using to show the UI?

Couple of other things you may or may not be aware of, the Pi only has 10/100 networking, so GigE makes absolutely no difference. Please don't use the on board 3.5mm jack for audio, its terrible! As I pointed out here https://github.com/volumio/Volumio2/issues/355 yesterday. Go and get yourself a DAC, it will be worth it.

ShanDestromp, 2016-04-14:
Chrome 49.0.2623.112 on two different Win7 stations; Chrome 51.0.2704.5 on Android 6; and whatever version of Midori is available in the official repos..

RE: GigE
I know; but I hardwired my entire house + workshop with GigE so that's what it's connected to.  The B+ was a spare I had lying around for this project; though I do plan on upgrading to a Pi3 at some point just for the extra processing power (even in a framebuffer the UI suffers quite a bit of lag).

RE: a DAC
I'm half deaf and this install is for when I'm in my workshop so the quality out of the 3.5 jack is adequate for my needs.  I honestly probably wouldn't be able to tell a difference anyways.  I'm using Volumio largely because it has a UI that works on a touchscreen without too much effort (neither Pi-Jukebox and Mopidy work right on my 7" screen).

apinner, 2016-04-14:
Fair points.

There are issues with the volume knob that we are aware of. I'm not sure if anyone has reported this yet though. It would be interesting to see it happening if you have the ability to show a short video?

volumio, 2016-04-14:
I'm reworking the backend volume mechanism a bit to improve the situation...

ShanDestromp, 2016-04-14:
[image]
Picture of my audio settings

https://www.youtube.com/watch?v=ygLQzVKJk3w Video of the volume jumping.  I'd like to point out that the actual volume level *is* changing; it's not just a display issue.  Video is from my desktop (Chrome, Win7); you can't hear the song because it's playing in my workshop.

apinner, 2016-04-15:
I haven't seen that happen before! That's very strange..

ShanDestromp, 2016-04-15:
So good news / meh news.  I swapped out my B+ for a Pi3 and *so far* the issue hasn't presented itself again.  I literally pulled the SDcard from my B+ and put it in the Pi3 (and moved the display over) so I'm *guessing* it's a performance issue not a programming one.

I do have a couple of critiques / UI enhancements for small-screen setups; not sure if you want me to make a new issue for that or just post it all here.

apinner, 2016-04-18:
@ShanDestromp If you have feedback for the UI, its probably best to create issues over at the UI repo. https://github.com/volumio/Volumio2-UI

apinner, 2016-05-17:
Volume has been rewritten and should be fixed in the next release.

closed 2016-05-17 by apinner

cjvanzwol, 2017-09-24:
It still happens on my setup. 

Rpi 2, Volumio 2.246, no DAC, official Volumio Android app as remote

messismore, 2017-10-07:
I have the same issue with a Zero + Hifiberry Amp+ using Safari on macOS 10.12 and iOS 11.