volumio/Volumio2 #361: [RC1] RPi 3 - audio output selection

rlucian, 2016-04-15:
on raspberry pi 3 , with RC1, if you boot volumio with the HDMI plugged in, volumio works fine but outputs the audio to hdmi only.

it would be useful to offer the option (by command line and through UI settings) to choose Audio-Jack, HDMI, or both. 

I think the other mediaplayer distro on raspberry has it like that.

apinner, 2016-04-15:
You are unable to go into Playback Options and select Audio Jack as the output device?

rlucian, 2016-04-15:
It's already selected, and it's the only option too, even when the HDMI is connected.
But when both are connected it goes to the hdmi, nothing on the audio-jack.

apinner, 2016-04-15:
What happens if you try?

`sudo amixer cset numid=3 <output>`

0=auto
1=headphones
2=hdmi

for 3.5mm jack

`sudo amixer cset numid=3 1`

rlucian, 2016-04-15:
for the moment i can only test it dry, as I am not in the office to actually hear if it worked, but running that remotely outputs:
```
# amixer cset numid=3 1
numid=3,iface=MIXER,name='PCM Playback Route'
  ; type=INTEGER,access=rw------,values=1,min=0,max=2,step=0
  : values=1
```
I will setup another pi3 at home tomorrow and test the output on the jack.

volumio, 2016-04-16:
Fixed with 
https://github.com/volumio/Volumio2/commit/221cdcfe5bf14c360cd6d4c83c66c688c2e027af

closed 2016-04-16 by volumio

rlucian, 2016-04-18:
```sudo amixer cset numid=3 1```
worked just fine, and properly switched the audio output to jack.
thank you.