volumio/Volumio2 #371: shutting down storage an option?

rlucian, 2016-04-20:
Would it be feasible (even manually) to disable the storage once Volumio is booted and up and running? Or does it need to write stuff to function?

Unmounting and maybe even turning off the power on the sd card could reduce power consumption and increase the life of the sd card, at the expense of not having logs, but there are many scenarios where this would really be worth it.
thanks.

apinner, 2016-04-20:
Volumio operates in RAM only, it doesn't constantly read/write to the SD card.

Realistically, how much power you going to save by turning off power to the SD card? We must be talking a few microamps...

For a device that probably costs about £5 to run for a year 24/7 in the UK, the saving is going be pennies. Or was you not thinking from a cost point of view?

rikardo1979, 2016-04-20:
you can install it on USB drive if you like and leave just small boot partition on SD card

rlucian, 2016-04-21:
Hey, thanks for the answers.

tl,dr;
The main concern is the card life, and it's not even about the cost, but the maintenance required if/when the card breaks.

My use-case is this: I am configuring volumio as managed digital-radio that checks-back with a central backend once in a while just to adjust stream source, volume, hostname, etc.
We deploy one or more in each retail store all across the country. Many of them are shut-off brutally (people but more often power losses) and this reduces their lifetime significantly. If I could at least unmount the card (power-down would be even better) power-loses wouldn't affect the card at all.

```umount /boot``` works just fine, but that's all i managed to do, the other targets are busy.
I know this is a linux question, but I was thinking that I might get some good pointers on this particular implementation so I can get to a point where I can execute a shell script after the system is up to turn off the sdcard.

Again, thanks for the help, and any hint is greatly appreciated. 

p.s. as soon as the RPi3 will receive the netboot (PXE) features I will use it and all my worries are gone, but this topic might still help others if I manage to find a solution. I am already using netboot on Intel NUC's to achieve the same result for in-store info-kiosks, but RPi3 would clearly be a much better alternative.

apinner, 2016-04-21:
It doesn't help your problem entirely. But we will be working on a feature that will download a complete backup of your volumio install including all settings that can then be reimported to a fresh install.

That would take away some of the pain should an SD card fail.

volumio, 2016-04-21:
SD Card durability has been a central design point from the beginning. 
What we do to extend its lifespan is:
- Ext4 without journaling
- File system is mounted on RAM and writes are written to a separate partition
- Read only core filesystem
- No logs are written to SD Card, but on RAM 
- IF there is corruption, it will be corrected at boot.

I'm quite confident that with this kind of attention, SD Cards will last YEARS over a 24\7 use...

PS: Send me an email at info at volumio.org , I may have something interesting to discuss with you..

rlucian, 2016-04-21:
Thanks for the prompt answer, you guys probably never sleep :+1: 

Just need to mention that setting up a new sdcard is not the big issue, it takes 5 minutes or less, but replacing the ones in remote locations is the real problem, as it takes time and effort. Having a backup sdcard ready in each location didn't proved a good idea, as they were either lost of already compromised by the time they were needed.

I do get your point though, and I am not really hoping that my request would become a feature, just looking for pointers so I can implement it when needed.

rlucian, 2016-04-21:
Yes, truth be told, I am running 4 instances since October on the stable release and not one had any issue this far, which is very encouraging.

Email sent, btw.

closed 2016-04-21 by apinner