volumio/Volumio2 #372: Missing title info on OLED

rikardo1979, 2016-04-20:
I use this kit mainly to stream HiFi music from Tidal. I do have some albums and tracks stored locally but most is coming from Tidal HiFi. I mean the digital if you say as otherwise I play vinyl.

And here is some issue which I was not able to solve just yet and I was not able to find any answers. When I stream from local network sources, all the Album-Track information appears on display. But when I stream from Tidal only few shows this on it. Most of the songs coming with time info only and "No song info" message.

Anyone have an idea where to start my troubleshooting? Is this due to Tidal issue(?), Bubble uPnP app which I use to send Tidal streams to Volumio, or is this down to Volumio itself?

So I just did a test and sent the same Tidal streams to our AVR which has streaming abilities and all titles appears on display normally
Than this issue would be down to Volumio and not Bubble remote app

Here is screenshot from webgui where you can see missing title from Tidal stream. Than on OLED display on streamer itself is just "No song info" message 

GUI screenshot https://i.snag.gy/r70Vya.jpg

apinner, 2016-04-20:
That's Volumio 1.55, this is for Volumio 2.

You will have more luck on the forum.

rikardo1979, 2016-04-20:
hmm, so what is the difference? I am a bit confused here

apinner, 2016-04-20:
Volumio 2 isn't officially released yet. It is a ground up rewrite that will supersede 1.55.

Download the RC1 version and give it a try.

rikardo1979, 2016-04-20:
I see. Trouble is I had some difficulties to make power button working and than OLED. So some guy on Audiophonics make an custom img with all scripts built in.

volumio, 2016-04-20:
Could you please post a link to this .img?

rikardo1979, 2016-04-20:
Forum (most in French, so not easy for me to read as I rely on google translator)
http://forum.audiophonics.fr/viewtopic.php?f=4&t=1492&start=75#p6200

direct DL from Mega
https://mega.nz/#F!QAUBlAxC!RyOJlP1OXbXH_gLvHSjADw

volumio, 2016-04-20:
Your best bet is probably to ask them for a Volumio2 based image...

rikardo1979, 2016-04-20:
hmm...they response times are ridiculous to be honest. Even I paid for their product the supposrt is close to none.
So I better use something else than Volumio. So going to find a different OS and start over
Thanks for reply here, anyway

volumio, 2016-04-20:
Sorry I don't get why you should use a different OS... 
However I've also had bad experiences with the Audiophonics guys... So I understand your frustration...

rikardo1979, 2016-04-20:
the reason being as there would be no support from this community here and lacking official support. So for this reason I think I should try something where I may get some support

apinner, 2016-04-20:
Lets be fair.. you bought a DIY audio kit from a company that isn't affiliated with volumio in any way. Their lack of support for their own product does not reflect on volumio at all and its unfair to imply otherwise.

Volumio is open source and provided for free, support is provided by the developers and the community where possible and for the vast majority it works. If you have something out of the ordinary provided by a third party volumio can't really be expected to support every single different variant.

closed 2016-04-20 by apinner

volumio, 2016-04-20:
Yes, however  I think he was not blaming Volumio in any way...

apinner, 2016-04-20:
ah, my mistake.. I must have read it wrong.

rikardo1979, 2016-04-21:
just don't understand your approach here but not very friendly you are...

volumio, 2016-04-21:
We don't need to. We develop this for free

rikardo1979, 2016-04-21:
it does not mean you have to be rude and offensive to who ever come here. We develop for free too but we treat every single individual with respect

volumio, 2016-04-21:
Sorry but I can't find any offense or unpoliteness here...

rikardo1979, 2016-04-21:
No problem. Good luck here

apinner, 2016-04-21:
I don't see where anyone has been rude?

I read your message as you blaming volumio for your equipment not working and because of volumio being unable to support the issue, you was going to go elsewhere and not use volumio any more.

To which I stated a couple of facts about your setup, volumio and how support is offered.

I then said I made a mistake and must have read your message incorrectly, as on a second read I don't think that was your intention.