volumio/Volumio2 #373: Set TimeZone

Alexqw, 2016-04-21:
Currently there's no way to set the timezone via the web UI. This is a problem when using the Alarm functionality (which, granted, doesn't work terribly well right now #319).

It's not a big deal for me to run `dpkg-reconfigure tzdata`, but I imagine the console would be off-putting to some users.

---Alex

volumio, 2016-04-21:
Yes I was thinking on it too... 
The deal is finding a list of timezones to display in a select field for the user to choose, then set localzone and reconfigure tzdata...
Any suggestion?

Alexqw, 2016-04-21:
I recommend using the data provided by the tzdata package. It lays out a folder structure in `/usr/share/zoneinfo/`. There's also a zone.tab file in there, which might be helpful, though it does suggest using a `zone1970.tab` instead, though that file doesn't exist... :-/

---Alex

apinner, 2016-04-21:
https://timezonedb.com/ nice API here

Alexqw, 2016-04-21:
@apinner To me, it doesn't make a lot of sense to use a remote service when the data is all available locally.

Furthermore, the local service is the one to be configured. The remote service and local tz database are likely (certainly?) to be updated at different intervals. This can lead to scenarios where input is used that the local `dpkg-reconfigure` won't understand.

---Alex

Alexqw, 2016-04-21:
@apinner Ahh, my bad about the remote service. I see you were recommending to download the database. My second point still stands though.

---Alex

volumio, 2016-04-21:
What do you think of this selector? 
http://mishguruorg.github.io/angular-timezone-selector/

The trick would be then translating into a bash shell script...

apinner, 2016-04-21:
Its OK, what is it like on mobile/touch?

apinner, 2016-04-21:
Actually works well on mobile, but the additional list sorting just makes the list endless and confusing!

Alexqw, 2016-04-21:
@volumio Internally, all `dpkg-reconfigure tzdata` actually does is create a symlink. For example

    ln -sf /usr/share/zoneinfo/Europe/Berlin /etc/localtime

So, just make sure your user input can be translated into that folder hierarchy, and it's pleasantly trivial. :-)

---Alex

volumio, 2016-04-21:
Yes, to make that more confortable we would actually need 2 selects, one for continent and the other for the region...

Alexqw, 2016-04-21:
It seems that all of the possible options for tzdata are in ``/var/lib/dpkg/info/tzdata.templates``.

Anyway, while not in the most convenient format, it looks to be easily parseable --- and is already localized.

---Alex

volumio, 2016-04-21:
Definetely too long... Is there a way to reconfigure the tzdata from the actual time of the timezone?

Alexqw, 2016-04-21:
@volumio What do you mean, like configuring it based off of what time it is right now in their locale? That won't work, as there's a lot of messed up locales in the world. Even in just the US, it would fail as many states don't follow DST.

---Alex

volumio, 2016-04-21:
No I just mean retrieving the time from the browser's time, and then setting this as locale by understanding where is in respect to UTC

Alexqw, 2016-04-21:
@volumio I don't know about getting the time from the browser, but still, all you've done is acquired the time, not the time zone. The time, even compared to UTC, is insufficient to determine the time zone. The easiest (though certainly not only) problem is that DST (or lack thereof) is not universal. Thus it would be off at some point in the year for anyone who is in a zone that borders with another zone that doesn't match their DST observation.

---Alex

Alexqw, 2016-04-21:
I found this project[1][2] (MIT licensed) that can determine the modern time zone in IANA format (which is the format used by the symlinks) via the browser.

Hopefully it is of some use.

---Alex

[1] https://bitbucket.org/pellepim/jstimezonedetect
[2] http://pellepim.bitbucket.org/jstz/

apinner, 2016-04-22:
Are we over complicating things?

janpieterz, 2016-04-25:
My 2 cents, the only difference I see in using the browser time and seeing the difference with UTC is that this isn't stable for locations where daylight savings time occurs. Having the actual time zone is very helpful for this. If you'd measure me currently, I'm at UTC + 2. In a couple of months I'll be back to UTC + 1, the alarm would be off by an hour, or know that it has changed (in which case this whole issue would become irrelevant because then local time simply is local time, which is also an [option](https://github.com/cdown/tzupdate)).

apinner, 2016-04-25:
https://www.youtube.com/watch?v=-5wpm-gesOY

I'll say no more...

volumio, 2016-04-25:
WOW!
Let's ban whoever pronounces timezones from now on... lol ! 

So, let's find the simplest way and put it into volumio, I'm quite puzzled at this point...

Fantastic channel BTW

xipmix, 2016-10-22:
Regarding the angular-timezone-selector, would it be possible to take the result it returns and set things like so:
```
tz_area=`echo $result | awk -F / '{print $1}'`; tz_zone=`echo $result | awk -F / '{print $2}'`;
echo tzdata tzdata/Areas          select $tz_area | sudo debconf-set-selections
echo tzdata tzdata/Areas/$tz_area select $tz_zone | sudo debconf-set-selections
echo tzdata tzdata/Zones/Etc:     select UTC      | sudo debconf-set-selections    #for completeness
sudo dpkg-reconfigure tzdata -ftype Noninteractive
```
I also noticed [moment-timezone](http://momentjs.com/timezone/docs/)

gogiman, 2018-01-16:
As mentioned by aqw,
ln -sf /usr/share/zoneinfo/Europe/Berlin /etc/localtime
works nicely.

macmpi, 2018-01-16:
consolidating all regional settings matters in [one thread](https://github.com/volumio/Volumio2-UI/issues/317)