volumio/Volumio2 #375: Support Password Protecting Web UI

Alexqw, 2016-04-21:
Sorry for the deluge of new tickets. ;-)

Node.js apparently doesn't support using Apache-style .htaccess files (useful for adding Basic Auth). I have no experience developing in node, but from what I read, there are only programmatic solutions for adding Basic Auth.

So, thus I opened this feature request, hoping for a simple password prompt in order to access the web UI.

---Alex

volumio, 2016-04-21:
Yes, it's on our to do list... It's a frontend issue anyway... We'll be using some angular or express middleware

Alexqw, 2016-04-21:
@volumio Excellent. :-)

Is this todo list available publicly? I didn't mean to open a bunch of tickets for things which are already known todos in a pre-release project. :-/

---Alex

fooness, 2017-02-28:
Dear developers, are there any news concerning this?

fooness, 2017-03-07:
I think this would be a real important feature. The last days I tried to look into the code, but I’m absolutely clueless on how to work with those websockets. Hope this feature will very soon exist.

Thank you in advance!

nyanloutre, 2017-03-21:
Would like to use this feature too, as my computer will be connected to a non trusted network (students residence)

This aside, awesome work :)

BeCracked, 2017-06-04:
I have the same issue (students residence) which is why I want to move on from the chromecast, which I am currently using, because it basicly broadcasts a "control me" to EVERY Android device in the network.
So requiring at least a simple password for accessing the frontend would make me feel a lot better, especially when you have logins openly visible in the frontend.

whatsnewsisyphus, 2017-10-07:
I need this too, similarly sharing a school network. Let me know if I could help in any way.

silelmot, 2018-04-18:
Is there smth new? I need this for a school too

alexnoyes, 2018-06-15:
Is there any update on this Feature Request?

pburndorfer, 2020-03-23:
Any updates concerning password protection?