volumio/Volumio2 #376: [RC1] Queue - webradios' names missing

THEJAJO, 2016-04-21:
When adding to the queue new webradio the name in Queue is empty. Sometimes, when double-click in Queue, it start playback, sometimes not.

apinner, 2016-04-22:
Not sure I really understand? 

You added a new webradio, when you add it to the queue it doesn't show in the queue? 

I've added a few web radios and I haven't seen this happen. What did you call the webradio?

THEJAJO, 2016-04-24:
I go to Browse | Webradio and add few internet radio stations to My Web Radios:
- Polskie Radio Czworka: mms://stream.polskieradio.pl/program4
- Off Radio Krakow: http://stream4.nadaje.com:13332/offradiokrakow
- TM-Radio: http://tm-radio.com/tmradio.m3u

All of them works fine when I double-click on any.
I add one by one to the queue and when I go to Queue some of their names are empty, some with different name.
When I double click on any radio in Queue, usually it works, playing, but sometimes not, showing Error - failed to decode, which is strange as the same Web Radio works fine when I double click it in My Web Radio list.

The other issue is when the Volumio2 on R Pi2 is online for a long time (24h or less) and not plying anything, it hangs and must be rebooted or even hard reset.

Thanks for help.

THEJAJO, 2016-04-25:
Another issue which totally make a mess.

I created two m3u playlists:
- http://thejajo.republika.pl/webradios/psyradio.fm.m3u
- http://thejajo.republika.pl/webradios/RMF.m3u

When I add them for a first time to My Web Radios and double-click on one of them, the radios included are added to the Queue and they can be played - still with empty names, however they works fine.

When I would like to add them to the queue, something strange is showing in the Queue (for example psyradio.fm.m3u) which is failing to decode.

The other strange thing is when I go back to My Web Radios, the listed radios has changed URLs, for the first ones from .m3u files. What the f... is going on? 0_o I need to re-edit the URLs to make it working again.

_Loading those URLs in foobar2000 works without any issue._

THEJAJO, 2016-05-01:
I have the same issue with .pls playlist - Webradio url is updated with first link from .pls file.

xipmix, 2017-01-30:
Is this still an issue for you, with 2.041? Quite a bit of work on webradio was done last year.

THEJAJO, 2017-01-31:
Yes, the issue was not solved.
None of the m3u/pls playlist are loading correctly.

I also noticed the names of radios are not updating - if there is any song change on the radio, there is no update in the name. On foobar2000 everything is working perfect.

shanehull, 2017-03-07:
This is an issue for me too. Almost every Shoutcast radio fails to decode.

Just sent through some bug reports.

closed 2017-06-02 by volumio