volumio/Volumio2 #385: Music on the SD card

ThomasTheGerman, 2016-04-28:
There should be a possibility to save music on the SD card that contains the system, for example by having a seperate partition for it that gets mounted to /mnt/LOCAL, in addition to the NAS and USB mounts.

At the moment, I have to use a workaround by editing fstab to mount my music partition to /mnt/USB.

volumio, 2016-04-28:
Yes, definetely. We'll add that

apinner, 2016-06-27:
@volumio was this added?

volumio, 2016-06-27:
Not yet, but I'll do

ThomasTheGerman, 2016-08-03:
@volumio 
This is listed in the features of RC2, but how do you do it? I just want to ask before I copy a few GB of music onto the SD card. Do you just put it in a folder into the volumio_data partition?

apinner, 2016-08-04:
@ThomasTheGerman you just browse to the IP of your volumio in a file explorer and there will be a folder called Internal Storage. Just copy your music there.

ThomasTheGerman, 2016-08-04:
@apinner That's slow as **** though, the transfer is bottlenecked by wifi and the write rate of the raspi. 

After transferring a couple of GB to it, I checked the SD card, but couldn't find the music (weirdly, the card did't show the space being used, but the music is there in the webinterface, even after reboot)

apinner, 2016-08-04:
Well the share is located /data/internal

Not much that can be done about the limitations of your rpi. How else was you expecting it to work?

nealcrook, 2017-01-06:
I found this whilst trying to store local data. Here's how I did it on my Pi Zero (volumio 2.041, December 2016)

1/ do a "first boot" from a clean image copied to the SD card
2/ connect to the hot-spot (log-in to wifi using password volumio2)
3/ navigate to volumio.local in a browser, reconfigure the network settings so that the Pi will connect to my usual home WiFi netowork
4/ connect back to the pi through its IP address on my local network (192.168.1.116 though yours will of course differ)

At this point I shut down the Pi, pulled the SD card and plugged it in to my Linux desktop. There are 3 partitions on the drive and the one you need to mount is "volumio_data". This is an ext<something> filesystem so windows users are out-of-luck with this approach - but see Plan B below.

Within the partion browse to dyn/data. There should be a directory named INTERNAL. if there is not, create it. Put your music in this folder. Umount the partition, pop the SD card back into the Pi and reboot. Viola, the "browse" area should show an INTERNAL folder which you can then scan to add to your collection.

Plan B is to connect to the running system. For example, using scp to copy stuff across.

If you ssh to the machine:

$ ssh volumio@192.168.1.116 
(password is volumio)

you can inspect the file system and see that /mnt/INTERNAL is a soft link to /data/INTERNAL (which is the dyn/data/INTERNAL directory referred to above)

So, you can use ssh to copy data across thus:
$ scp -rp tori_amos volumio@192.168.1.116:/data/INTERNAL

Hope that helps someone.

Neal.

closed 2017-06-02 by volumio