volumio/Volumio2 #387: Airplay streaming from Synology Audio Station

Turbofoehn, 2016-04-30:
Volumio 2 no longer plays audio streamed from Synology's Audio Statio app via Airplay.
The play head in the app jumps forward (apparently randomly) and there will be audio stutters for fractions of a second here and there, so Volumio is definitely receiving and decoding. Volume adjustments work as well.

Anybody with a Synology DiskStation able to reproduce this? It used to work flawlessly with 1.55. Streaming to Airport Expresses works, so I don't think Audio Station is to blame here, streaming to Volumio with iTunes works as well, so it's not a bandwidth problem either…

closed 2017-06-02 by volumio

jpbaril, 2018-03-10:
I'm having the same problem. Any solution?

messismore, 2018-03-11:
Are you on the latest versions of both Audio Station (6.4.0-3313) and Volumio (2.368)?

Works for me now…

jpbaril, 2018-03-11:
Audio Station yes, but Volumio I am not as I'm using it on a Cubox-i and last version is 2.129.