volumio/Volumio2 #392: Selecting the DAC on an OdroidC2 needs a reboot

gkkpch, 2016-05-03:
The RC1 version needs a reboot after selecting the DAC as the output device. Failing to do that results in a plop and noise when starting a track. After reboot /etc/modules holds both snd_soc_pcm5102 and snd_soc_odroid_dac, Volumio starts OK. 
Perhaps not both modprobes are done when selecting the DAC? Could also be it works as designed?
Anyway, easy to reproduce for someone with a C2 ( @volumio :) ) and low prio.

codewarriorFX, 2016-12-21:
Hi, was the DAC connected via I2S or a USB controller?

gkkpch, 2016-12-21:
ooops, should have closed this one ages ago as this has been resolved.
But to answer your question, yes, it was connected to i2s.
Connecting an usb audio DAC usually works on the fly as far as I know.

closed 2016-12-21 by gkkpch

codewarriorFX, 2016-12-21:
LoL, I posted very recently on the Issues about connecting an PCM5102 via I2S. But the dac is asynchronous. Maybe you can help me?

gkkpch, 2016-12-21:
Saw it, not looked at it yet. Will do...