volumio/Volumio2 #396: [RC1][WebUI] Static DHCP doesn't work

THEJAJO, 2016-05-04:
I'm not able to set static IP.

When I try to set up, it shows it was successful, but when I go back to the Network setting it has still DHCP enabled. The static IP address which was set before doesn't work.

Raspberry Pi 2

apinner, 2016-05-04:
I think this may be a duplicate of https://github.com/volumio/Volumio2/issues/337 which has been resolved but not in a release yet.

apinner, 2016-05-05:
If this is a different issue to 337 then please reopen.

THEJAJO, 2016-05-05:
In this case I'm writing about LAN, not WLAN, and changing the setting using WebUI (web browser), not console connection.

If I have permissions, I will re-open the issue.

apinner, 2016-05-17:
Can confirm that this doesn't work.

Error while setting network: TypeError: fs.creatOutputStream is not a function.

apinner, 2016-06-27:
This is now resolved in RC2

closed 2016-06-27 by apinner