volumio/Volumio2 #399: Access to SMB Share

gkkpch, 2016-05-04:
"Access to network drive (password-protected SMB-Share) is lost after reboot" has been reported by an Odroid C2 user.
I also have encountered the same thing occasionally and not paid real attention to it (re-entered user and password and saved, problem solved).
This only happens rarely. Network connection/ timing related?

I did not find anything similar in the open issues, so perhaps this should be investigated.

volumio, 2016-05-04:
I saw this issue as well, so added the noauto mount option which mounts the share the first time its accessed, this should avoid try to mounting the NAS when no network has been established...

Now we're a bit "detached" since the system has made a considerable step up, but it's all in dev branch... We still need to finish some things before merging and making a new image... 
You can have an idea by using the dev build branch... Let me know what you think

closed 2016-05-05 by gkkpch

gkkpch, 2016-05-05:
Pulling/ Building right now and will test it...

gkkpch, 2016-05-05:
Not very usable yet I'm afraid, but a potential winner!!!
Very nice to see plugins can be installed, a really great leap forward.
Spotify plugin does not allow to store user details yet and hence it won't work.

gkkpch, 2016-05-05:
Actually, I don't get any sound out of it since adding spotify...