volumio/Volumio2 #410: Unable to load MPD playlist

ivesdebruycker, 2016-05-11:
MPD playlists (/var/lib/mpd/playlists/) are shown in the root of the music library, but when trying to play them, an error is thrown:
```
info: [1462956944404] ControllerMpd::sendMpdCommandArray
info: Error:
Error:  [50@1] {add} Not found
    at MpdClient.receive (/volumio/node_modules/mpd/index.js:57:17)
    at Socket.<anonymous> (/volumio/node_modules/mpd/index.js:37:12)
    at emitOne (events.js:90:13)
    at Socket.emit (events.js:182:7)
    at readableAddChunk (_stream_readable.js:147:16)
    at Socket.Readable.push (_stream_readable.js:111:10)
    at TCP.onread (net.js:523:20)
```

The cause is that `ControllerMpd.addPlay` is called with the name of the playlist (without extension), so command 'add' is sent instead of 'load' (/volumio/node_modules/mpd/index.js:145).

I don't know a reliable way of knowing if is in fact an MPD playlist, except maybe `!fileName.startsWith('NAS/')`.

Perhaps these playlists should also better be located in 'Playlists' instead of 'Music library'?

volumio, 2016-05-11:
Since we're using our own playlist method, didn't think to integrate that... But we could add m3u\pls handler after RC2 release...

ivesdebruycker, 2016-05-11:
I was thinking about a plugin that would read m3u-files from a folder and converts them (periodically) to volumio2-playlists, but first i'll need to have a look at the plugin structure.

volumio, 2016-05-11:
Good idea... We're finalizing it, hopefully this weekend will be out

xipmix, 2016-10-22:
This may be fixed now, can you give 2.001 a try?