volumio/Volumio2 #419: [RC1] playback stopped after error "mpd[447]: db: No such directory"

firemeteorxx, 2016-05-15:
From the journal:

> May 15 06:29:54 volumio volumio[414]: info: Pushing Favourites {"service":"mpd","uri":"/NAS/Nas/gregorian - love songs/10 - when i need you.mp3","favourite":false}
> May 15 06:30:00 volumio volumio[414]: info: [1463293800662] Play playlist Enya
> May 15 06:30:00 volumio volumio[414]: info: CoreCommandRouter::volumioClearQueue
> May 15 06:30:00 volumio volumio[414]: info: [1463293800694] CoreStateMachine::ClearQueue
> May 15 06:30:00 volumio volumio[414]: info: [1463293800701] CorePlayQueue::clearPlayQueue
> May 15 06:30:00 volumio volumio[414]: info: CoreCommandRouter::volumioPushQueue
> May 15 06:30:00 volumio volumio[414]: info: [1463293800716] InterfaceWebUI::pushQueue
> May 15 06:30:00 volumio volumio[414]: info: CoreCommandRouter::executeOnPlugin: mpd , clearAddPlayTracks
> May 15 06:30:00 volumio volumio[414]: info: [1463293800737] ControllerMpd::clearAddPlayTracks
> May 15 06:30:00 volumio volumio[414]: info: [1463293800746] ControllerMpd::sendMpdCommandArray
> May 15 06:30:00 volumio mpd[447]: player: played "NAS/Nas/gregorian - love songs/10 - when i need you.mp3"
> **May 15 06:30:00 volumio mpd[447]: db: No such directory**

I'm not sure if the complaint is about this specific music file or something else.
I can confirm that the music file and the directory exists. It has played first 9 files from the same directory without problem and died on the 10th. Upon this failure it clears the play queue instead of skipping to the next song.

firemeteorxx, 2016-05-18:
Some more info on this issue.
It appears that the 'No such directory' log does not happen on switching to a new song.
It could happen half way during playback.
What's more, in some more serious situation, this could lead to dead-loop in node.js and the system does not play any new song without a reboot.

When dead-loop happens, the index.js process grabs CPU forever and send a lot of logs to journal.
(BTW, I have to change the journalctl config to persistent to get journal visible, hope this does not harm.)
  415 volumio   20   0  162656  44424  16996 R 76.9  9.0  10:30.40 /usr/local/bin/node /volumio/index.js                                                                                      
  138 root      20   0   12480   4700   4388 S 15.1  1.0   3:00.72 /lib/systemd/systemd-journald                                                                                              
  684 volumio   20   0  120508  27072  13600 S  3.2  5.5   2:28.60 /usr/local/bin/node /volumio/app/plugins/system_controllers/system/telnetServer.js

apinner, 2016-05-18:
One for you @volumio

firemeteorxx, 2016-05-25:
I have more information on this bug.
It turns out that it has nothing to do with play-back, but a regular recurring issue.
Playback will get interrupted if issue shows up during playback.
See the following log, it appears that some task is triggered every hour and causing the fail.
May 25 07:30:00 volumio mpd[446]: db: No such directory
May 25 08:30:00 volumio mpd[446]: db: No such directory
May 25 09:30:00 volumio mpd[446]: db: No such directory
May 25 10:30:00 volumio mpd[446]: db: No such directory

Is my DB corrupted?

apinner, 2016-05-25:
I would imagine it is the same issue as this https://github.com/volumio/Volumio2/issues/433

volumio, 2016-05-25:
No, it's a different issue... Which device do you have?

firemeteorxx, 2016-05-27:
I'm on raspberry pi zero with hifiberry DAC pro. The network is through a wifi dongle. The music is hosted on nas mounted through NFS.

After reporting the last update I tried to refresh the DB and unfortunately messed up everything. What I did is to remove the mount and trigger a rescan. It does not seems to work. So I tried to add the same mount back. Something must have went wrong since that. I now loss connection to the box after a while, haven't got chance to invest yet.

firemeteorxx, 2016-05-27:
Btw, I'm now no longer able to play anything

firemeteorxx, 2016-05-27:
Can anybody provide a custom build of MPD with the following change to provide some debug context?
SimpleDatabasePlugin.cxx-        error.Format(db_domain, DB_NOT_FOUND,
SimpleDatabasePlugin.cxx:                     "No such directory: %s", selection.uri.c_str());

I couldn't set up the cross compile env...

firemeteorxx, 2016-05-28:
The error log generated from debug build:
May 28 08:30:00 volumio mpd[439]: db: No such directory: NAS/Nas/Enya 1988-2009.17CD/Enya.1988.Watermark/CDImage.cue

The path is NOT a valid path in my current setup, due to my recent change in Music lib organization.
But I think it used to be a valid path.
The thing is that I don't know where this comes from.
I think I have deleted the old /var/lib/mpd/tag_cache... Why it still refers to the old path?

closed 2017-06-02 by volumio