volumio/Volumio2 #421: UI stuck on playback

hfatgit, 2016-05-16:
UI is coming up and then is stuck with the "hourglass" blocking every other action

I am using a PI B+ with Volumio 2 RC1

hfatgit, 2016-05-16:
2016-05-16T18:19:32.028Z - info: -------------------------------------------
2016-05-16T18:19:32.081Z - info: -----            Volumio2              ----
2016-05-16T18:19:32.089Z - info: -------------------------------------------
2016-05-16T18:19:32.095Z - info: -----          System startup          ----
2016-05-16T18:19:32.101Z - info: -------------------------------------------
2016-05-16T18:19:32.282Z - info: Loading plugins from folder /volumio/app/plugins/
2016-05-16T18:19:32.336Z - info: Loading plugins from folder /data/plugins/
2016-05-16T18:19:32.355Z - info: Plugin system is not enabled
2016-05-16T18:19:32.363Z - info: Plugin i2s_dacs is not enabled
2016-05-16T18:19:32.370Z - info: Plugin network is not enabled
2016-05-16T18:19:32.377Z - info: Plugin networkfs is not enabled
2016-05-16T18:19:32.385Z - info: Plugin services is not enabled
2016-05-16T18:19:32.392Z - info: Plugin alsa_controller is not enabled
2016-05-16T18:19:32.399Z - info: Plugin volumiodiscovery is not enabled
2016-05-16T18:19:32.407Z - info: Plugin upnp is not enabled
2016-05-16T18:19:32.414Z - info: Plugin airplay_emulation is not enabled
2016-05-16T18:19:32.421Z - info: Plugin dirble is not enabled
2016-05-16T18:19:32.433Z - info: Plugin alarm-clock is not enabled
2016-05-16T18:19:32.441Z - info: Plugin albumart is not enabled
2016-05-16T18:19:32.448Z - info: Plugin example_plugin is not enabled
2016-05-16T18:19:32.455Z - info: Plugin mpd is not enabled
2016-05-16T18:19:32.462Z - info: Plugin spop is not enabled
2016-05-16T18:19:32.469Z - info: Plugin updater_comm is not enabled
2016-05-16T18:19:32.476Z - info: Plugin mpdemulation is not enabled
2016-05-16T18:19:32.484Z - info: Plugin websocket is not enabled
2016-05-16T18:19:32.581Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
2016-05-16T18:19:32.592Z - error: Plugin alsa_controller is not loaded. Unable to get an instance
2016-05-16T18:19:32.598Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
2016-05-16T18:19:32.604Z - error: Plugin alsa_controller is not loaded. Unable to get an instance
2016-05-16T18:19:32.611Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
2016-05-16T18:19:32.617Z - error: Plugin alsa_controller is not loaded. Unable to get an instance
2016-05-16T18:19:32.623Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
2016-05-16T18:19:32.626Z - error: Plugin alsa_controller is not loaded. Unable to get an instance
2016-05-16T18:19:32.688Z - info: BOOT COMPLETED
2016-05-16T18:19:32.893Z - info: [1463422772886] CoreStateMachine::resetVolumioState
2016-05-16T18:19:32.902Z - info: [1463422772896] CoreStateMachine::getcurrentVolume
2016-05-16T18:19:32.909Z - info: CoreCommandRouter::volumioRetrievevolume
2016-05-16T18:19:32.971Z - info: [1463422772964] CoreStateMachine::updateTrackBlock
2016-05-16T18:19:32.982Z - info: [1463422772974] CorePlayQueue::getTrackBlock
2016-05-16T18:19:33.712Z - info: [1463422773698] VolumeController:: Volume=undefined Mute =undefined
2016-05-16T18:19:33.737Z - info: [1463422773731] CoreStateMachine::updateVolumeundefined
2016-05-16T18:19:33.747Z - info: [1463422773741] CoreStateMachine::pushState
2016-05-16T18:19:33.757Z - info: [1463422773751] CoreStateMachine::getState
2016-05-16T18:19:33.767Z - info: CoreCommandRouter::volumioPushState
2016-05-16T18:19:33.773Z - info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
2016-05-16T18:19:33.779Z - error: Plugin volumiodiscovery is not loaded. Unable to get an instance
2016-05-16T18:19:33.812Z - error: Plugin websocket is not loaded. Unable to get an instance

apinner, 2016-05-17:
This is after a reboot?

hfatgit, 2016-05-17:
After reboot, after killing the node.js process and restarting it. Always the same.  Volumio UI starts and then runs into the picture above

apinner, 2016-05-17:
Have you tried reflashing?

hfatgit, 2016-05-17:
Three times. Always the same

volumio, 2016-05-17:
Ok, do this:

rm -rf /data/configuration
killall node 

It should start normally after that.. Let me know

hfatgit, 2016-05-17:
Yess.  
Looking great. 
UI started well. 
Can't wait to test it

Thank you for your help

closed 2016-05-17 by hfatgit