volumio/Volumio2 #424: Library missing songs/artists

Saiyato, 2016-05-18:
I have found that some songs/artists won't show up in the library.

I think there are two issues at hand here, one has an error in the log:

1. id3: Failed to read RIFF chunk > this should be resolved in mpd 0.19.x (I have updated the mpd client, but to no avail, maybe I'm doing something wrong here? I used MPD 0.19.9) > link to issue: https://bugs.musicpd.org/view.php?id=4486 

2. no error found in log, but for example the directory 'Alizée' was not indexed, maybe because of the 'é' in the directory name? If I list the directory (SSH) is shows with a question mark on the place of the 'é', but I can browse it and see all contents.

In both cases the music is also indexed in iTunes and has artwork, problem 1 is a ripped CD with standard artwork from iTunes, problem 2 (in this case) has custom (self provided) artwork, still added using iTunes.

I love to help and/or debug if you don't have the time for it. I kinda do need some directions though.

Please let me know if anything is unclear or if I need to provide additional information.

Setup:

Raspberry Pi B2/B+/B (have tried all three)
Volumio 1.55 and 0.86 (Volumio 2) image
Music is on an NFS-share (Windows Server 2012R2)
Network connection is wired
No upgrade/dist-upgrade/rpi-update has been performed

apinner, 2016-05-20:
Out of interest, what happens if you rename that folder but copy and paste the é letter from say here https://en.wiktionary.org/wiki/%C3%89#Translingual

I did some testing a while back where i created a load of folders and files with id3 tags that just contained random characters like that and i had no issues at all. Just wondering if its an encoding issue.

Saiyato, 2016-05-22:
Good morning,

 

My apologies fort he late reply, but I have been kept busy (family affairs).

I will tend tot he below early coming week. Is there any log I might need to check?

 

Met vriendelijke groet // Kind regards // Mit freundlichen Grüssen // Cordialement,

 

Rachid Groeneveld

Saiyato, 2016-05-23:
I have tested a bit, it seems Volumio has some issues and does not reboot correctly all the time. Will try and search the issues whether this has been submitted yet. I have made a picture of the screen it hangs on.

I have created an NFS share with two folders, one album for each problem.

Problem 1:
No change, the album is Trash (Alice Cooper).
[image]

Problem 2:
Replacing the character with visually the same one did not help. Maybe the problem exists in reading Windows from Windows NFS shares, because when I change the directory to Alizee, it indexes all songs. It even reads the accent aigu and artwork (submitted it myself in iTunes) from the ID3 tag.

I will try testing the above again against an SMB instead of NFS.

Can I check any logfiles, or is indexing not logged by default?

Saiyato, 2016-05-23:
Problem one occurs on both CIFS and NFS shares.

Update for problem 2, when adding a CIFS share as opposed to NFS the directory with the accent aigu was picked up instantly. No problem whatsoever, so the problem probably lies within the reading of directory names using NFS (and probably other encoding).

I can see some notifications when scanning the library using the NFS share:
[image]

The difference in the SSH is that the NFS share shows the directory with one question mark: Aliz?e, and the CIFS share shows two question marks: Aliz??e.

Saiyato, 2016-07-19:
I believe the problem lies within MPD and not volumio. After some trial and error I recompiled mpd (version 0.19.17) and the problem was gone. This applies to all versions of Volumio, a newly compiled mpd solves these problems (version 0.19.13+ should suffice).

closed 2016-07-19 by Saiyato